CTCaer/hekate #236: [Enhancement] - add function to format, partition microSD, and inject nand images to emuMMC Manager

3xkrazy, 2019-06-30:
Perhaps this may be a useful feature for advanced users. Something that's similar and popular in the 3ds scene is d0k3's GodeMode9 homebrew as seen here:
https://github.com/d0k3/GodMode9

deurk, 2019-06-30:
```
Nyx and hekate TODO
In no particular order:

Sd raw partition backup
Migration from raw to sd and the opposite
Clock offset support like HOS
Temporary emuMMC disable
Destructive SD partitioning
Resize based SD partitioning
emuMMC resizing
Sd raw emuMMC on unallocated empty space support?
Quick launch of last chosen boot entry
Themes support
Joycon support
```

Part of changelog notes

3xkrazy, 2019-06-30:
Oh, I didnt notice that part.

closed 2019-06-30 by 3xkrazy