CTCaer/hekate #237: Hekate don't boot OFW 6.0.1

UdoChudo, 2019-06-30:
So, next problem. When try boot to OFW 6.0.1 through Hekate 5.0 it just return me on Hekate
OFW 6.0.1
Hekate 5.0
EmuMMC 6.0.1
SD Card Fat32 partition for games

Hekate_ipl.ini
> [config]
> autoboot=1
> autoboot_list=0
> bootwait=3
> customlogo=1
> verification=2
> backlight=40
> autohosoff=0
> logopath=bootloader/bootlogo.bmp
> { }
> [Atmosphere]
> payload=bootloader/payloads/fusee-primary.bin
> { }
> [Stock 6.2.0 and lower]
> stock=1
> emummc_force_disable=1
> { }

CTCaer, 2019-06-30:
Hmm I'll need to investigate that. Did it work in 4.10.1?

Slluxx, 2019-06-30:
Same for me with a completly clean 6.1 system

bbatod, 2019-06-30:
Yeah, I encountered an issue with 6.2 as well. Switch FW 6.2 + Atmosphere 0.9.2 + Hekata 5.0 and just gets stuck on boot for me.

Ultracoolguy, 2019-07-01:
For me, if I use fusee-primary.bin(while pretty slow due to me using File-based EmuMMC) it boots into EmuMMC, so you guys can use that as a workaround.

bbatod, 2019-07-01:
> 
> 
> For me, if I use fusee-primary.bin(while pretty slow due to me using File-based EmuMMC) it boots into EmuMMC(6.2 btw), so you guys can use that as a workaround.

I did not even setup emuMMC, I was just trying to boot into CFW on 6.2 with Atmosphere 0.9.2 and Hekate 5.0. Reverting back everything to Atmosphere 0.8.9 and Hekata 4.10.1. fixed it.

Ultracoolguy, 2019-07-01:
For me it looks like, EmuMMC or no EmuMMC, 6.*(and potentially lower) doesn't work.

In my case my SysMMC is on 8.0.1 so I'll stick with 5.0.0 until then.

UdoChudo, 2019-07-01:
> For me it looks like, EmuMMC or no EmuMMC, 6.*(and potentially lower) doesn't work.
> 
> In my case my SysMMC is on 8.0.1 so I'll stick with hekate 5.0.0 until then.

on SysNand 4.1.0 emuMMC with Hekate 5.0 and Atmo 9.2 working good. Don't tested 5.0.1 FW but i think it will work too. Problem in 6.X only but not sure

CTCaer, 2019-07-01:
So hekate 4.10.1 + atmo 0.9.2 works with 6.x?

Ultracoolguy, 2019-07-01:
Yep, and latest fusee-primary works fine too.

CTCaer, 2019-07-01:
Good, I might pinpoint it then.

CTCaer, 2019-07-02:
Various stuff were fixed. I have a build in retronx discord.
Please check.

Ultracoolguy, 2019-07-02:
Will try it ASAP.

Ultracoolguy, 2019-07-03:
It may be me only, but while I don't get stuck on splash screen now, when I boot with Hekate in Stock to 6.2.0, after the splash I get a black screen. The system is rebooting to RCM(I proved it by connecting my phone while it rebooted. It booted back to hekate).

For curiosity, I decided to try CFW to see if it would boot, and it got past the Nintendo Switch logo, but after that Atmosphere gave me an error with code 2002-4686. The title points to the ns module, according to SwitchBrew.

CTCaer, 2019-07-03:
That error is probably the NS patching for emuMMC.

Do you by any chance run stock with emuMMC? This is not supported, as it needs a fix on m4's emuMMC

Ultracoolguy, 2019-07-03:
I used to, but I switched to OFW SysMMC(6.2.0 for now) and CFW emuMMC(8.0.1). I don't have any problems with emuMMC, this is in sysMMC. I ran Atmosphere in 6.2 because I thought it could give us a clue. Sorry if I didn't make it clear(English isn't my primary language).

CTCaer, 2019-07-03:
6.2.0 was fixed internally.

But I'm quite curious about 6.1.0/6.0.x.
I wonder if the problem is the debug fix patches.

To test that, I'll need someone that really is on 6.x that is not 6.2 and want's to test stock with no emuMMC. (As I said before emuMMC with stock doesn't work and will be removed until fixed. It will throw an error for that purpose.)

Ultracoolguy, 2019-07-03:
I know you want people lower than 6.2, but it may not be working at all. Like I mentioned in one of my comments, booting 6.2 on Stock in sysMMC redirects to RCM. Doing the same, but with CFW goes fine until the ns error(the same goes to fusee-primary). Sorry if I'm being stubborn, it's just that it may not be fixed yet(and at the same time I'm worried that my backup may be corrupted).

CTCaer, 2019-07-03:
Can you show me your ini which boots to Stock 6.2?

Ultracoolguy, 2019-07-03:
Sure(I renamed it to a .txt file since Github doesn't accept inis.
[hekate_ipl.ini.txt](https://github.com/CTCaer/hekate/files/3354969/hekate_ipl.ini.txt)

Ultracoolguy, 2019-07-03:
Well crap, I accidentally disabled autoRCM and now I don't have hekate. 

Good news though, it isn't hekate. My switch threw the same error it threw with Atmosphere. Any ideas on what should I do?

CTCaer, 2019-07-03:
Your stock option run emuMMC. Which is not working.
You need to use 

```ini
[Stock]
fss0=atmosphere/fusee-secondary.bin
stock=1
emummc_force_disable=1
icon=bootloader/res/icon_switch.bmp
```

Ultracoolguy, 2019-07-03:
Well, I'll try once my jig arrives(which according to Ebay, in 10 days). 

Still, I know this isn't related to hekate, but why does booting without hekate to Stock throw the same error as with Atmosphere? Is my backup screwed up?

CTCaer, 2019-07-03:
dunno what error is that

Ultracoolguy, 2019-07-03:
I meant the error that pointed to the ns module. It's the same error code(2002-4686), although I don't know if it's the same module from last error, due to Stock's limited error message. This is now happening on Stock without hekate(because I accidentally disabled autoRCM).

CTCaer, 2019-07-03:
Invalid FAT format for BIS User partition.

Ultracoolguy, 2019-07-03:
So, what should I do?

CTCaer, 2019-07-03:
It has to do with your USER partition. You probably did sth that you shouldn't have.

Ultracoolguy, 2019-07-03:
I don't care if I lose my saves in sysMMC since I already backed them up. I should boot into Recovery Mode and there I should reinitialize the console, correct?

Ultracoolguy, 2019-07-03:
Well, the initializing provides the same error. I'm gonna ask on reddit in order to keep this more on-topic

CTCaer, 2019-07-06:
https://github.com/CTCaer/hekate/commit/0862cb1e7e7b9d9a46edebd55436762ac45c8801

closed 2019-07-06 by CTCaer