CTCaer/hekate #241: How to generate the res.pak for nyx?

friedkeenan, 2019-06-30:
I've been looking in the repo and I don't see the files for it, though I have been able to dump some images from it by looking at the code. It does seem that a lot of stuff for it is hardcoded, so maybe instead of generating it, it should at least be file on the repo?

CTCaer, 2019-06-30:
Well I actually forgot to add it. So I will.
I hand made that, because I didn't want to create any header or else and go into making a creator, which I don't have the time to do.

friedkeenan, 2019-06-30:
Sounds good, and thanks for all the hard work you've put into this

friedkeenan, 2019-07-02:
I don't know if this is useful/what you want, but I made this: https://github.com/friedkeenan/nyx-res-tool

closed 2019-07-06 by friedkeenan