CTCaer/hekate #243: Booting both stock and CFW hangs on 5.1.0

friedkeenan, 2019-06-30:
Happens even without nyx, and when looking at the debug output it seems to hang after [this line](https://github.com/CTCaer/hekate/blob/d55634638f35c30b24620ee0a5be09c2bbbcb75b/bootloader/hos/hos.c#L317). I get the hang using release builds and my own builds.  I somehow got it to work last night, but I took out my SD card this morning to see what would happen when injecting hekate without it, and then it started hanging again.

ckurtz22, 2019-07-01:
Having the same issue on 4.1.0

ckurtz22, 2019-07-01:
So the trick I found is to add a crap ton of gfx_printf() calls in that area of the code until it boots fine, then just remove all of em and it'll still work :shrekate:

CTCaer, 2019-07-01:
I suspect why it does that. And I believe it's stack corruption.

The codegen changed a lot because of some stuff being Os and most being O2.
Can you try removing all ini_free and ini_section_free and try to break it?

CTCaer, 2019-07-01:
Btw I have a build in retronx server for that. I removed some unneeded frees that could corrupt the stack.

CTCaer, 2019-07-02:
@friedkeenan this was fixed. Can you try a build in retronx?

I'm trying to get as more stuff ready as I can now.

closed 2019-07-06 by friedkeenan