CTCaer/hekate #245: Can't boot Lakka

NickHam13, 2019-07-01:
Whenever I extract the L4T Lakka release to my SD card and then try to boot it through hekate 5.0.0, it shows the hekate splash screen, then the screen turns off for a couple seconds, then the fan starts really high for a second and then it hangs on a black screen. I've made sure everything is set up correctly beforehand, and I've used the exact same Lakka setup on the same SD card.

chillyshackk, 2019-07-01:
Sounds like the same issue most of us had with L4T Lakka, pretty sure it's not related to hekate but the way you formatted your card.Solution was to first create a msdos partition table before creating the actual partitions. I was able to boot lakka using 5.0.0 and latest atmos etc...

NickHam13, 2019-07-01:
Sorry if I sound like a complete idiot, but what's an MS-DOS partition table?

chillyshackk, 2019-07-01:
I am no dev or expert on the subject but after messing around trying to get it to work that's what made a difference for me and a couple others. 

I use linux so if you're on windows you'll need to find a partition app that can create partition tables, maybe windows does it by itself, maybe it has the option but I no longer have windows installed so I cannot check. 

And if you're on linux, see my post here:
https://gbatemp.net/threads/l4t-lakka-switch-now-with-audio-and-vulkan.538619/page-36#post-8651595

NickHam13, 2019-07-01:
I don't think this is the case because Lakka worked 100% of the time on the same SD card.

chillyshackk, 2019-07-01:
Strange, got mine to work earlier, it was a copy of my previous lakka folder and not a clean install though.

chillyshackk, 2019-07-01:
So after retesting it was no longer working, so what I tried was restarting the whole setup except that I did not create the emuMMC and it works so it seems that it stops working after creating the emuMMC. Since I have multiple sd cards I dedicated one to lakka and have emuMMC and homebrews on another one.

NickHam13, 2019-07-01:
I tried a smaller SD card with just hekate and Lakka and it worked. So I guess this is more of an emuMMC issue. Do you wanna make the issue or should I?

closed 2019-07-01 by NickHam13

chillyshackk, 2019-07-01:
Go ahead :)