CTCaer/hekate #246: Can't make second emuMMC

davidv3rs, 2019-07-01:
I can't seem to make a second emuMMC. I have three partitions total, a main FAT32 and two RAW partitions for emuMMC. The create button manages to create a 1st emuMMC just fine, but I can't switch to the second SD RAW 2 or create another emuMMC using the "Create" button...it just re-creates emuMMC/RAW1.

Is this a known issue?

CTCaer, 2019-07-01:
Yeah it always shows the first available.
This is not implemented yet.

LeMageFro, 2019-07-02:
A quick workaround for now is to start by creating the first partition at the end of your SD with the space for the second partition still unalocated before, then use hekate to make it an emuMMC, then create the second partition before the first one, and finally use hekate again to make this new partition a second emuMMC. After that the "change emuMMC" option works just fine to switch between the two.

CTCaer, 2020-05-01:
Added in version

closed 2020-05-01 by CTCaer