CTCaer/hekate #247: Failing to load nyx?

N3MIS15, 2019-07-01:
Launching hekate stalls at the boot screen (Custom NX bootloader...) when i have sd card inserted. Launching without an sd card inserted launches the "old" style menu where i can then launch other payloads etc. Launching with autoboot enabled also works as intended.

I have tried cards formated with both fat32 and exfat, launching the payload from switchme chip, reboot to rcm and also launcing the payload again from within hekate itself. I also tried cards from different manufacturers.

 Please let me know if you require any other information as I am out of ideas of what to try to get this working.

FWIW It works just fine on my 4.1 ipatched switch via caffine.

CTCaer, 2019-07-01:
I wonder what your device doesn't like. I'll think of sth.

chip1, 2019-07-01:
I've the same problem. Tried to Update OFW from 6.2, to 7.01 and 8.1 without success

Geranospiza, 2019-07-01:
This happens to me as well. I mean, its not only the backup. Also fails to load CFW (Atmosphere). I've seen the github issue. Im on 8.1.0. I have 2 micro SD cards that have successfully backed up my system on hekate 4.10.1, no issues. None of them work in 5.0.0 One is a samsung U3, the other one SanDisc U1.

[image]

CTCaer, 2019-07-01:
@chip1 @Geranospiza 
Please don't mix issues with booting with issues in loading nyx. They are completely separate things.
Problems with booting are probably a stack corruption thing, while loading Nyx, needs more solid investigation (although I have an idea).

And I hope I will see some of you in discord so you can try test builds

N3MIS15, 2019-07-01:
Which discord server?

CTCaer, 2019-07-01:
I'm only in one server, retronx

chip1, 2019-07-01:
@CTCaer: I mean the nyx boot issue, Stuck on bootscreen. Tried also different SD cards with a fresh and clean configuration. Older versions and with classic GUI everything is running well.

CTCaer, 2019-07-01:
I added some specific wait times that were missing. Please check in discord.

CTCaer, 2019-07-01:
The issue is newer models of touch panels

Geranospiza, 2019-07-01:
This is not true. I had my switch one day before oficial release and Nyx
hangs on me.

CTCaer, 2019-07-01:
hmm can you test a build then in retronx discord?

Geranospiza, 2019-07-01:
Of course. Will be home in 2 hours. DM me the build. Im Rogers, the dude
with the 512gb samsung evo plus U3 card.

CTCaer, 2019-07-01:
they are attached there. you just need to rename them to nyx.bin.

chip1, 2019-07-02:
@CTCaer: the new build gives a black screen after bootlogo.

CTCaer, 2019-07-03:
Turned out the problem was different POR values for different SoC revisions.

N3MIS15, 2019-07-03:
Confirmed fixed.

CTCaer, 2019-07-03:
This will be closed when the commit that has the fix gets commited.

Webbeh, 2019-07-03:
Hi.

I'm not sure whether the issue is the same or not, but I've got a XAJ1001**** Switch, with firmware 8.1.0.

Can send the payload just fine, works with both the GUI and the old style menus...

...until I create an emuMMC (RAW partition). After that, injecting hekate with SD card in freezes on the Nyx' boot screen (Custom NX bootloader). Injecting hekate without SD card displays the old style menu.

I tried both with and without this fix : https://github.com/CTCaer/hekate/issues/235#issuecomment-507036263 

Also, before creating emuMMC, the "Launch" menu displayed all the options (Atmosphere, ...), and after creating an emuMMC (without powering off or removing the SD card), it said that there is no config file available.
Going on the SD Card information screen showed the available space on the card, as expected.

Is this related to OP's issue ?

CTCaer, 2019-07-06:
This was fixed by various commits, but most cases were this: https://github.com/CTCaer/hekate/commit/5b919cb12e4bbd33c81790611277597939e3983f

closed 2019-07-06 by CTCaer