CTCaer/hekate #250: Stock emuMMC black screen

davidv3rs, 2019-07-01:
Stock emuMMC does not boot into Horizon. It results in a black screen. OFW is 4.1.0.

RetroGamer74, 2019-07-01:
There is an Atmosphere fix in 0.9.2 updated from today for that issue. Re-download it.
Also add the flag force_emmumc_disable=1 in your stock ini file.

davidv3rs, 2019-07-01:
> There is an Atmosphere fix in 0.9.2 updated from today for that issue. Re-download it.
> Also add the flag
> 
> emummc_force_disable=1
> 
> in your stock ini file.

Yeah, but I want to use stock emuMMC. That flag will disable emuMMC all together...

davidv3rs, 2019-07-01:
I also read somewhere that I need custom secmon and warmboot binaries. I compiled exosphere and put flagged exosphere.bin and lp0fw.bin but it still doesn't want to boot.

mexxs, 2019-07-01:
I think I have the same issue. I tried restoring my clean 4.1.0 backup to emuMMC partition on the SD card just to test it before I restore it to the actual eMMC. Restoring process worked fine. But I can only boot into emuMMC with CFW. Stock emuMMC gives me a black Screen after the Kosmos logo. 
My OFW firmware is 4.1.0. And I did all of this with the updated version of AMS 0.9.2. 
Hope this helps. Thank you for your hard work! Nyx is awesome! :)

LeMageFro, 2019-07-01:
I have a similar issue but on a different firware, I can't boot emuMMC stock 8.0.1, even though emuMMC CFW 8.0.1 boots just fine. My sysnand is 8.0.1 and it boots on stock without any issues also. The readme says that not all firmwares are compatible with stock emuMMC booting, do we know more about which ones are and aren't supported ?

CTCaer, 2019-07-01:
I have a test build in retronx server, that might fix that.

davidv3rs, 2019-07-01:
> I have a test build in retronx server, that might fix that.

Sweet, thanks! Is it already merged?

CTCaer, 2019-07-01:
nope, because I don't know if it works. It's a testing build

CTCaer, 2019-07-02:
So emuMMC currently does not work on stock.
I'll leave the issue as reference and remove the ability, until it's fixed.

This issue should be directed to the emuMMC repo.

CTCaer, 2019-07-06:
https://github.com/CTCaer/hekate/commit/c40c125462d20cda55e69168027a45045aeec18c

For progress on that, check https://github.com/m4xw/emuMMC

closed 2019-07-06 by CTCaer