CTCaer/hekate #251: SysNAND boot stuck at "loaded warmboot and secmon" after restore

UnicornsOnLSD, 2019-07-01:
I've set up an EmuNAND that works fine. I restored my SysNAND and am trying to boot it. I removed Nyx to see the log (Nyx stays on bootlogo) The backup was on 6.2 and I have an acceptable amount of fuses (and I'm using AutoRCM). The backup was from when I first got the Switch. Here is the output when I boot stock:
Initializing...

```
Found pkg1 ('20181107105733')
Identified pkg1 and Keyblob 6
Loaded pkg1 & keyblob
Decrypted & unpacked pkg1
Patching Warmboot
Patching Security Monitor
Loaded warmboot and secmon
```
Here is my hekate_ipl.ini:
```
[config]
autoboot=0
autoboot_list=0
bootwait=1
verification=1
backlight=100
autohosoff=0
autonogc=1

{AtlasNX/Kosmos v13.0.1}
{}
{Discord: discord.teamatlasnx.com}
{Github: git.teamatlasnx.com}
{Patreon: patreon.teamatlasnx.com}
{Pegascape DNS: pegascape.sdsetup.com}
{}

{--- Custom Firmware ---}
[CFW (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
{}

[CFW (EMUNAND)]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
{}

{--- Stock ---}
[Stock (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1
icon=bootloader/res/icon_switch.bmp
{}

[Stock (EMUNAND)]
fss0=atmosphere/fusee-secondary.bin
stock=1
icon=bootloader/res/icon_switch.bmp
{}
```
I'm trying to launch Stock (SYSNAND)

CTCaer, 2019-07-01:
you hanged at pkg2 decryption.
That means that you forgot to restore BOOT0/1.

UnicornsOnLSD, 2019-07-01:
That worked! Thanks :)

closed 2019-07-01 by UnicornsOnLSD