CTCaer/hekate #252: add multilanguage suport

ImanCol, 2019-07-01:
please, add other lenguage from spanish to Nyx

UdoChudo, 2019-07-01:
Maybe and Russian too :D

wetor, 2019-07-02:
I've written the font generation tool (pak file) and I think it's usable, but my compiled NYX doesn't work properly, so I can't make my language version. If the hekate team hadn't planned to support multiple languages, I might have done it (provided, of course, that I could run my compilation)

wetor, 2019-07-02:
My compilation results
hekate.bin  126,911 bytes (don't work)
nyx.bin  239,489 bytes (don't work)
*.bso file no problem

devkitARM is the latest version

wetor, 2019-07-02:
@CTCaer

CTCaer, 2019-07-02:
@wetor 
It's because of gcc9. The files are bigger. I made some clean up yesterday and trying to fix other issues.

About multi-language support:
This was always my intention, since lvgl supports that. BUT for that to begin, nyx must reach a certain level of stability and also have important missing features in. After that I can work on a special solution for that. (Currently if multilanguage support is enabled, the payload is too big)

wetor, 2019-07-02:
Thank you for your reply . I am looking forward to the follow-up update. :D

wetor, 2019-07-02:
> @wetor
> It's because of gcc9. The files are bigger. I made some clean up yesterday and trying to fix other issues.
>
> About multi-language support:
> This was always my intention, since lvgl supports [...]

Thank you for your reply . I am looking forward to the follow-up update. :D

Anxietytimmy, 2021-04-06:
i could translate nyx to spanish, but thats about it

impeeza, 2021-05-07:
Count on me for Latin American Spanish translation :D

jshsakura, 2022-01-24:
I always wanted to do Korean language support.
As long as it's in json or text format, I'd like to help the project too.

ImanCol, 2022-01-24:
there is no need to create a special version with texts in Spanish.  My intention was that they add that language if it supports mutilanguage.  If having multiple languages ​​can make the Payload bigger, just a plain text bootloader with a single custom language would suffice (oh well take the id the NAND is set to and take the corresponding language file in the bootloader folder.

mktm235, 2024-02-29:
Regarding multilingual support, if you want to support it in a format such as json, the way I think it is, you will need to check the master json file and make changes every time you add, delete, or change messages. I would like to actively cooperate, but I am worried that it will affect the hekate team's coding.
It is difficult to support multilingual payloads (size limitations), so I think it will only be supported by nyx.
If nyx has not reached a "certain level" yet, proceed with the fork.

ImanCol, 2024-03-11:
Some day

closed 2024-03-11 by ImanCol