CTCaer/hekate #254: explore sd card

ImanCol, 2019-07-02:
I would like to have the possibility that in Hekate + Nyx there would be a file manager. I do not say it for pleasure, or for suggesting it, but because I wanted to eliminate (or modify) the CFW configuration files related to the KIP load, this is because when I activated a KIP In Kosmos ToolBox (Use Kosmos .. .) when starting the CFW I get an error. Well I would like to use it only in case of emergency like when you delete a custom Theme and you do not want to start the Nintendo Switch, so to remove the Title that causes the problem from Hekate + Nyx would save me a lot of loading MemLoader + SDCard Command or extracting the SDCard.

CTCaer, 2019-07-02:
It was planned initially. Got scraped for emuMMC manage. It will come eventually. I still have the code at side.

CTCaer, 2019-12-08:
There's code here that can be possibly used for that
https://github.com/CTCaer/hekate/pull/317/commits/b91e48c999ab8211fb19c14d5067907b543e2733#diff-a15cc7dd475cd046b56bea55bd82bc58

juju35, 2022-01-29:
This is feature is still planned or not ?

impeeza, 2022-01-30:
It's planned. By now you can use the UMS tool on NYX and connect the switch as a USB stick on your computer

Visualwit, 2024-02-03:
An eMMC/emuMMC explorer would also be great!

ImanCol, 2024-02-03:
This was not implemented, but there are alternatives

closed 2024-02-03 by ImanCol