CTCaer/hekate #257: Lakka L4T messes with Temperature display

staccmang, 2019-07-02:
Hekate version: 5.0.0

Issue:
booting into lakka l4t (version 20190515), then rebooting into rcm and pushing the hekate payload again temporarily makes the temperature display at the bottom display °F values (the indicator itself still says °C, but the value is ~101, which corresponds to about 38°C)

the switch itself is not hot at all and the issue occurs even when you just boot lakka and immediately reboot into rcm/hekate. 

powering off from hekate fixes the issue.

CTCaer, 2019-07-02:
Hmm, it probably configures it to have that. I'll check it

CTCaer, 2019-07-04:
@staccmang 
Is it possible for you to test the one in retronx server?
I added some init to reset specific values.

staccmang, 2019-07-06:
sure, i'd be happy to help,
i just got into all of the switch stuff recently and don't really know any of the resources, could you provide a link?
or are we talking about the retronx discord?

CTCaer, 2019-07-06:
https://github.com/CTCaer/hekate/commit/fed15a2f2f56a2babbcbc0eccd906e623c86375f

closed 2019-07-06 by CTCaer