CTCaer/hekate #260: Can't create emuMMC from backup

belleseba, 2019-07-03:
(Similar to #246)
"Create emuMMC" button has a description "You can create it from eMMC or a eMMC Backup" but I guess currently there is no way to select "create it from a eMMC Backup"

Unintentionally started creating process takes a very long time and very frustrating because it cannot be canceled (right?).
If this is "Not yet implemented" status, I'd like to know how soon it's planned.
Then I can understand I have to do it manually for now if not so soon.

CTCaer, 2019-07-03:
Oh there is, but no there.

Migrate allows you to convert a backup

-  from backup folder to sd file based.
- Restore WITH Sd raw partition **ON** allows you to restore a backup from restore folder to sd partition.

and yes you can't cancel the creation process because I don't have code that handles it sanely yet.

belleseba, 2019-07-03:
Thank you very much!
I'm currently trying to work with two SD raw partitions.
So, "Migrate" doesn't support conversion from backup folder to SD raw partition based yet?

I have just understood backup files need to be in "restore" folder, started restoring and now I'm waiting for finishing it.

CTCaer, 2019-07-03:
`backup/{eMMC S/N}/Restore` folder for RAW.
And you use Tools->Restore -> Toggle `SD emuMMC Raw Partition` to **ON** ->Boot0/1 and Rawnand

`backup/{eMMC S/N}/` folder for file based.
And you use Migrate.

For restore to RAW, you need to have the button `SD emuMMC Raw Partition` **ON**

belleseba, 2019-07-04:
OK, thank you!
It seems I have succeeded to create the partitions and make both emunands work.
Personally now I have finished the setup, may only need to use "Change EmuMMC" function by Nyx.
Anyway this is really nice.  Thank you very much again for developing this.

closed 2019-07-04 by CTCaer

kristal91, 2020-05-09:
hi is it possible to create a beckaup from an emmc that I have on the sd and restore it on the original nand? I broke the nand and I don't have the beckaup automatically so it would be beckaup from the emmc I have and restore it