CTCaer/hekate #262: Add warning to clear archive bits on nintendo folder

fennectech, 2019-07-07:
It is highly recommended NOT to clear archive bits on the nintendo folder if you have a fat32 filesystem.  This will render all games over 4gb unbootable and you will have to re install them.  This is because the switch uses the archive bit for the NCAs stored for installed titles.  This information isnt communicated to the user.

CTCaer, 2019-07-07:
It clears the arc bit from everything and only adds it for .nca folders.

https://github.com/CTCaer/hekate/blob/master/nyx/nyx_gui/frontend/gui_tools.c#L208

closed 2019-07-07 by CTCaer

fennectech, 2019-07-07:
Oh. Thats cool.  👍    Didn’t know you handled that :)

CTCaer, 2019-07-07:
https://github.com/CTCaer/hekate/commit/0d9aebda7bc1225806375350e075cacf6956158f

fennectech, 2019-07-07:
Could we set archive bit on all directories containing a “00”.   This will get all files intended to be split that way.