CTCaer/hekate #266: screen  touch

Hodgeli, 2019-07-09:
hello，My switch screen is out of touch（I don't know why）,but I find it work all right in the Hekate 5.0.1.  what is your magic ? Is there any chance for me to save my screen touch?    my switch system is 6.0.1.  thanks a lot.

BaamAlex, 2019-07-09:
Delete or move the nyx.bin from /bootloader/sys. And you will now have the old layout with button navigation.

Hodgeli, 2019-07-09:
no。。。I mean my screen touch is work on well in the Hekate 5.0.1.   
but it is can not work in system and game . How amazing!!!
whatever ,thanks

Hodgeli, 2019-07-09:
> Delete or move the nyx.bin from /bootloader/sys. And you will now have the old layout with button navigation.

no。。。I mean my screen touch is work on well in the Hekate 5.0.1.
but it is can not work in system and game . How amazing!!!
whatever ,thanks

CTCaer, 2019-07-09:
So you can't use it at all even in Switch home menu / settings, but you can in hekate.

Hmm that's interesting.

You can use it in Lakka or L4T ubuntu?

Hodgeli, 2019-07-10:
> So you can't use it at all in Switch home menu / settings, but you can in hekate.
> 
> Hmm that's interesting.
> 
> You can use it in Lakka or L4T ubuntu?

yeah !!!  my screen touch is all right in Lakka!!!    Dose it means my touch is ok , the problem is comes from system driven?  if i upgrade or degrade my system version , is there any chance to save my screen touch??  haha

CTCaer, 2019-07-10:
Upgrading HOS (switch OS) may help. It's a very weird bug to exist on official software (except if you replaced your touchscreen)

closed 2019-07-28 by CTCaer