CTCaer/hekate #267: Custom Background

BaamAlex, 2019-07-09:
I want to know how i can set up an custom background for the new nyx gui.

NickHam13, 2019-07-09:
Take a 1280x720 image and convert it to 32-color bitmap. Make sure it's 32-color or it won't work. Then name it background.bmp and drop it in bootloader/res.

BaamAlex, 2019-07-09:
Tyvm. Should i edit the .ini for that?

BaamAlex, 2019-07-09:
Ehm...i tried it. But it won't work...

CTCaer, 2019-07-09:
The alpha is taken into account. If you used photoshop or an old GIMP it will come out with 0% alpha, which means 100% transparent.

You can use an online tool that allows you to convert it to 32bit RGB (where A becomes xFF)

NickHam13, 2019-07-09:
This one is alright:

https://online-converting.com/image/convert2bmp/

CTCaer, 2019-07-09:
Select `32 (True color, RGB)` first and THEN drag n drop your image.

closed 2019-07-28 by CTCaer

steinvm, 2026-02-15:
hey hallo ik krijg het maar niet voor elkaar om de achtergrond in hekate te veranderen .
ik heb eerst de afbeelding bewerkt naar 1280 x 720 ,72 en dan geconverteerd naar bmp
dan naam veranderd in background.bmp en in de juiste res file gekopieerd en relaod maar nog steeds geen succes
kan iemand me helpen aub greetz