CTCaer/hekate #268: Warn user on emuNAND creation

Luro02, 2019-07-09:
As someone, who just wrote the emuMMC to the main partition (the one with all the files) of the sd card, I propose to add a warning for the user, that you have to create a separate partition or you will corrupt your sd card! (or even better: prevent hekate from writing to partitions with files/folders on it)

Strat00s, 2019-07-09:
Maybe show what is on the partition that you chose.

CTCaer, 2019-07-09:
Can you give steps to reproduce?
The checks start at partition 1 and not 0 which is the main.
Also it says which partition chose. It says [Part #] and below the partition table.

Luro02, 2019-07-09:
Never mind, I just saw, that this is no longer possible :+1:

closed 2019-07-09 by Luro02

CTCaer, 2019-07-09:
Hmm, it wasn't possible even before.
That's why I asked for possible steps to reproduce