CTCaer/hekate #270: 4.0.1 restored stock (burnt fuses), pkg2 decryption failed

kankuro666, 2019-07-10:
I have restored my Boot0/1 and rawnand that I've made around a year ago on 4.0.1 - totally clean.
I did some CFW stuff, but now I restored (but by mistake i restored to 6.1 and disabled autoRCM, so now i have 7 burnt fuses).

I restored to my clean NAND, but it won't boot normally, so I wanted to use hekate, but I get the error "pkg2 decryption failed".

Is my NAND corrupted, or is hekate not yet supporting 4.x stock boot? Please help :(

kankuro666, 2019-07-10:
I used hekate 5.0.1 text one, in the graphical one after choosing stock - it is stuck at "HEKATE - IPL custom nx bootloader"

kankuro666, 2019-07-10:
[image]
[image]

Here are images

CTCaer, 2019-07-10:
You forgot to restore BOOT0/1.
Currently they are the 6.1 ones.

kankuro666, 2019-07-10:
Thanks a lot for your answer!
I'm pretty sure I restored BOOT0/1 multiple times, unless... i somehow overwritten my boot files with 6.1 ones in my 4.0.1 folder...

Do you think, that doing the procedure of:
1. restoring 6.1
2. downgrading to 4.0.1 using choi
3. backing up BOOT0/1
4. restoring 4.0.1 rawnand and also the new boot0/1

Would be a proper workaround? Or do I miss something.

fennectech, 2019-07-10:
This is a viable option. You can also downgrade to 4.0.1 with Choi and then
restore your nand backup.

fennectech, 2019-07-10:
Downgrading with choidujorNX to 4.0.1 will get boot0 and boot1 into the proper state for your nand backup.  Do keep a copy of your boot0 and boot1.   You dont need to restore them if you downgrade with Choi tho.

kankuro666, 2019-07-10:
That's a great idea! I will get back to you after I'm done with the procedure

kankuro666, 2019-07-11:
Thank you all it worked. I updated officially and I am not even banned ! Perfect !

closed 2019-07-11 by kankuro666