CTCaer/hekate #272: Savegame extraction?

Clydefrosch22, 2019-07-24:
Sorry if this is actually not feasible from a technical point of view, but would it be possible to use hecate to back up just the savegame data from the consoles nand? I'm specifically thinking for extracting saves from a clean, online ready system to inject on a hacked console with a savegame manager. 

Currently, it appears that extracting the files requires quite a bit of work and a full nand backup (i.e. https://gbatemp.net/threads/edit-ofw-clean-switch-save-data-from-nand-backup-restoring-via-fusee-gelee-payloads.541081/) and I wonder if that could be streamlined from the new and improved nyx gui.

Its a low priority addition for sure but I think it would be handy to have.

CTCaer, 2019-07-28:
This will never be a priority for hekate.

But the possibility is there as a hekate module or even a payload.

NickHam13, 2019-07-28:
I'd love to see this.

CTCaer, 2019-12-08:
Closing for now.

I'm prep'ing for this special module support.
Maybe a dev will find interest into implementing that

closed 2019-12-08 by CTCaer