CTCaer/hekate #273: failed to init sd card with 5.0.1

kuaner, 2019-07-25:
sd card init failed with 5.0.1 but the same sd card works great whit 5.0

CTCaer, 2019-07-25:
Can you fetch a photo of `Console info -> SD card / microSD info` from a working hekate version?

Clydefrosch22, 2019-07-27:
it's possible that your sd card/card slot are just a little wonky

whenever i take my card out, it takes a couple attempts before the switch feels like recognizing it again. possibly, the reader is lose or the connector has some issue, or the contacts on the card itself are damaged/scratched in too much.

CTCaer, 2019-07-28:
@kuaner any news about that info?

kuaner, 2019-07-29:
[image]
[image]

i use 5.0.0 as payload to init the bootup, then it works great with  5.0.1

that's very strange

CTCaer, 2019-07-29:
This can happen with fake sd cards.

Your card uses a valid id/oem/model but the year/size are wrong. And I doubt it's a V30 card.
What speeds you get for read/write with usb3 UHS-I sd card reader?

kuaner, 2019-07-30:
It's a 64 GB sd card
50MB read speed
80MB write speed

I would glad to try the payload

CTCaer, 2019-09-17:
I added some possible fixes for your issues. Please test again.

Otherwise, ping me at retronx discord for a special testing build that has some big SDMMC driver changes that will come out with v5.1.0.

CTCaer, 2019-12-08:
I didn't have any contact in discord, so I will mark it as solved.
Feel free to reopen if not.

closed 2019-12-08 by CTCaer