CTCaer/hekate #274: Touch not working on Nyx

paink1ller64, 2019-07-26:
Touch not working on Nyx, but after loading OFW or Atmosphere works perfectly

CTCaer, 2019-07-28:
After testing, it seems that this specific touch panel needs extra configuration for mapping the area correctly.

[image]

This will need investigation via the full STMFTS driver.

r1cola, 2019-07-28:
interesting...are there actually different models/SKUs of the panels in circulation?

paink1ller64, 2019-07-29:
found some information about it:

"this is some info from china.
nintendo switches are manufactured with two different models of touchscreen.
one is "NFT-K12D C2", AKA"K12D";
another is "E226116".
These two types of touchsceen are non-interchangable with each other (and I am so sure about that).
and AFAIK, the basic hardware drivers are diffrent."

CTCaer, 2019-07-29:
Well there are 4 actually. Touch IC remains the same. Same goes for the drivers (they cover the different cases, based on calibration info).
Only touch panels should change.
The fw sent is what probably defines the API that the panel must speak to IC.

MasterDriverRex, 2019-08-04:
Just here to say that Hekate v.5.0.0 and v.5.0.1 don't work for me; just reformatted my microSD card, downloaded everything again from sdsetup.com (so everything should be up-to-date), and have been on OFW 8.10 for ages on an unpatched console I bought in September 2018.

The top four options (Home, Tools, Console Info, Options) work fine enough, but anything below them doesn't work _at all_, no matter how I try to tap or push them. I don't know what kind of touchscreen I have, but when I boot into OFW or CFW using Hekate 4.10.1, the touchscreen works perfectly well. My screen isn't dirty either, due to me using a glass screen protector and rarely playing undocked. 

If this isn't the place to post this, then I apologise.

MasterDriverRex, 2019-08-04:
If I could humbly suggest something, could you please fork this and add emuMMC to 4.10 (the one controlled via power and volume buttons instead of touch), or allow buttons to be used? I have no idea whether either idea's feasible, but they should provide workarounds for those like me and paink1ller64 who are having issues.

fennectech, 2019-08-05:
If you delete res.pak it will revert to the old text interface for hekate @MasterDriverRex

MasterDriverRex, 2019-08-06:
@fennectech When I delete res.pak and use TegraRcmGUI to upload hekate_ctcaer_5.0.1.bin, I don't get the old text interface; I get the new interface, but with the text and images scrambled. 

[image]

fennectech, 2019-08-06:
Woah. Thats trippy ill take a peek at what needs to be changed to disable nyx.

NickHam13, 2019-08-06:
nyx.bin needs to be deleted to go back to the old interface.

fennectech, 2019-08-06:
Was just about to reply with this.  You beat me to it :)

fennectech, 2019-08-06:
@CTCaer  Can you satisfy a curiosity of mine and explain what’s causing the neat patterns?   Garbage in the buffer?  If its just random garbage why does it seem to have a pattern to it.

NickHam13, 2019-08-06:
@fennectech My theory is that Nyx is attempting to read null data because it's looking for res.pak, however it can't find anything so it just reads a bunch of 1's and 0's or whatever is occupying the space.

m4xw, 2019-08-06:
Undefined behavior

NickHam13, 2019-08-06:
I wish I knew how to program so I could understand what that means.

fennectech, 2019-08-06:
It simply means behavior that is not intentionally created by the developer.   Generally its a bug.

NickHam13, 2019-08-06:
Ah, gotcha. That makes sense.

fennectech, 2019-08-06:
I’m just wondering WHY it has a pattern instaid of random noise

NickHam13, 2019-08-06:
I wonder why everything retains the same color it's supposed to be too. Maybe that color data is stored in nyx.bin?

fennectech, 2019-08-06:
From how text works the color is specified within the program.   The font / image data is monochromatic.  You’ll notice anywhere the same letter is is going to be identical.  (But lets take this to discord.  AtlasNX?

MasterDriverRex, 2019-08-07:
Anyway, yeah, deleting nyx.bin did bring back the old interface; thank you for that, @NickHam13 and @fennectech. Still unfortunate how emuMMC is restricted to Nyx, when Nyx doesn't work at all on my system, but oh well.

NickHam13, 2019-08-07:
You can still set up emuMMC without Nyx. I'm pretty sure you need to compile it yourself at the emuMMC repo, extract everything to the right place, combine your boot0, boot1 and rawnand.bin, then dd it to a partition on your SD card. It should work the same.

fennectech, 2019-08-07:
I’m pretty sure you dont need to compile emummc yourself. Its just a mater of setting up the config properly

NickHam13, 2019-08-07:
Yeah, I think everything required actually comes with hekate. So he would need to flash his boot0, boot1 and rawnand.bin all combined into a single file to a partition on his SD card and set up his config file.

MasterDriverRex, 2019-08-07:
@NickHam13 Which config file?

CTCaer, 2019-08-07:
@MasterDriverRex Creating emuMMC from hekate main is not possible, because it doesn't fit in the binary. My mitigation for touchscreen like these, will be at start, a joycon driver.
>Which config file?

emuMMC/emummc.ini

---------

About the fonts/icons, it has that pattern because when RAM powers on, it doesn't have all bits as 1 or 0. They are randomly populated. There's always a MBIST that writes values for testing the ram.

Anyway let's not derail the touch panel convo here.

vassall89, 2019-08-14:
I have the same problem , i use hekate to boot Android then 2 days after that when i try to use hekate the touch screen doesn't work properly , just can press in some points , but on sysnand everything works fine , hope to see controller support soon.

CTCaer, 2019-08-15:
@vassall89 Oh, that's peculiar.
So you could use touch before.

What changed inside these 2 days? For example, did you update HOS?

vassall89, 2019-08-15:
Well nothing happened in the next 2 days after trying Android... In fact i tried Android for 6 hours rebooted the console and used hekate for the second time because the playstore wasn't working... so it worked 2 times for me before stop register touch in the entire middle section of the screen,Sorry for my ignorance but what is HOS? I pretty sure didn't update nothing after tried Android!

fennectech, 2019-08-15:
HOS Is horizon. The nintendo switch’s operating system.  (The one nintendo
made and is on all switches
-- 
FennecTECH

MasterDriverRex, 2019-08-15:
> Sorry for my ignorance but what is HOS?
Official firmware.

CTCaer, 2019-08-29:
So next version will have Joy-Con support as a fallback for these issues

taufikismailn, 2019-09-13:
> I have the same problem , i used hekate to boot Android then 2 days after that when i try to use hekate the touch screen didn't work properly , just can press in some points , but on sysnand everything works fine , hope to see controller support soon.

same thing happen to me. i have try at least 10 times, the top menu sometimes not working too, the sub menu always not working

MasterDriverRex, 2019-09-14:
Does Nyx 0.8.2 have Joy-Con support? I don't see it listed on the release page, is all...

CTCaer, 2019-09-15:
@MasterDriverRex Next bigger version will (5.1.0).

CTCaer, 2019-09-17:
Just an update on this matter. It seems that newer touch panel firmware from HOS causes that issue. It may be possible to reset that state, but what I tried till now, it didn't work.

taufikismailn, 2019-09-18:
> > I have the same problem , i used hekate to boot Android then 2 days after that when i try to use hekate the touch screen didn't work properly , just can press in some points , but on sysnand everything works fine , hope to see controller support soon.
> 
> same thing happen to me, touch screen work well before. i have try at least 10 times, the top menu sometimes not working too, the sub menu always not working

Fix on 5.0.2

CTCaer, 2019-09-18:
@taufikismailn huh? 5.0.2 fixed the issue for you?

taufikismailn, 2019-09-28:
> @taufikismailn huh? 5.0.2 fixed the issue for you?

Yes, Touchscreen on hekate 5.0.2 now working perfectly for me

paink1ller64, 2019-09-28:
I tried hekate 5.0.2, nothing changes for me

CTCaer, 2019-09-29:
@paink1ller64 @taufikismailn 
HOS version? Asking this because 9.0.0 seems to have updated touch firmwares.

paink1ller64, 2019-09-29:
> @paink1ller64 @taufikismailn
> HOS version? Asking this because 9.0.0 seems to have updated touch firmwares.

8.1.0

C-P-O, 2019-10-11:
Hi same issue here, touch screen not working with Hekate, having same issue with SX OS
I also changed the Touch Screen for a new one, still not working, only top button are clickable.
Current OS 9.0.1

taufikismailn, 2019-10-13:
> @paink1ller64 @taufikismailn
> HOS version? Asking this because 9.0.0 seems to have updated touch firmwares.

7.0.1 sysnand when the problem occur, i update the hekate 5.0.2 (from atlas kosmos) then it fix it by itself. Then I create emunand 7.0.1, 8.1 and 9.0.1 with sysnand still on 7.0.1 and touchscreen still working till now
 
When the problem happen, i've try to fix it by using exfat/fat32, use hekate from here and kosmos, restore ofw 7.0.1, using different card, but neither fix it

Sorry for late reply

maskmark, 2019-10-19:
[image]
Maybe there's alternative workaround of menu navigation with touch button mapped in the effective screen area. no need for touch panel mapping info or hekate joycon driver.

fennectech, 2019-10-19:
i doubt joycons will ever work in hekate.

MasterDriverRex, 2019-10-31:
Nyx used to work for 9.0.0 and 9.0.1, but I've updated to 9.0.2 and the latest Kosmos (via KosmosUpdater app), only for it to stop working.

Sigh...

CTCaer, 2019-10-31:
9.0.2?

MasterDriverRex, 2019-10-31:
Sorry, 9.0.1.
Nyx used to work on 9.0.0 when I manually put all the homebrew files (from SDSetup) on my microSD card; now, on 9.0.1 (not 9.0.2), Nyx is again unresponsive to touch inputs.

Reply to: CTCaer/hekate <reply@reply.github.com>
Date: Thursday, 31 October 2019 at 10:15 pm
To: CTCaer/hekate <hekate@noreply.github.com>

CTCaer, 2019-10-31:
Yeah that's caused by hos. If you are willing, can you try these sensitivity options in HOS settings?

I'll also try some new touch commands these days and see if it fixes it.

MasterDriverRex, 2019-10-31:
Wait, what "sensitivity options"?

CTCaer, 2019-10-31:
There are options in Settings -> Controllers and sensors.

MasterDriverRex, 2019-10-31:
[image]

That's odd...there's a band on the left side where it suddenly jumps further to the left than I'm touching...tested it with both 'Standard' sensitivity and 'Stylus' sensitivity, and that band's still there. Also, the left border of that rectangle? It resets whatever I drew, as if I had lifted my finger up, even when I'm going in circles around it.

[Touchscreen Sensitivity Issues.zip](https://github.com/CTCaer/hekate/files/3795714/Touchscreen.Sensitivity.Issues.zip)

(Had to take 8-second videos on my phone, transfer them to laptop via OneDrive, export the .mp4 files to 480p, then compress the two .mov files into that 5.5 MB .zip to upload - GitHub doesn't support .mp4 or .mov or files > 10 MB)

r1cola, 2019-10-31:
that's odd, but you never noticed any issues with the touch in those areas in actual games though?
and this was a launch unit?

fennectech, 2019-10-31:
This is a hardware fault.  Call nintendo mate

r1cola, 2019-10-31:
> 
> 
> This is a hardware fault. Call nintendo mate

...or not, hehe
likely they would be able to tell that unit's "illicit" history ;)

fennectech, 2019-10-31:
Well. if you use choidujorNX to do a downgrade while factory resetting it.  then updating it with no autorcm and booting stock. it will line everything up.  Youll prolly be just fine

fennectech, 2019-10-31:
But.  youll likely get back a F-G patched unit.  so. thats one down side.  You can send it to a 3rd party repair place

MasterDriverRex, 2019-10-31:
Not too familiar with choidujorNX, and I haven't used autoRCM in a while.

Besides, I mostly my Switch in docked mode - for better visuals, and to use my Wii U Pro Controller via Magic-NS (which is a much more comfortable controller than the official two options) - so the band doesn't really bother me much.

MasterDriverRex, 2019-11-01:
Anyway, doesn't most of the homebrew stuff live on the microSD card? What would Nintendo be able to find on my system memory, if I were to send my console in? Lastly, wouldn't a reformat ("Restore Factory Settings") clear out leftover homebrew in system memory?

[69 more replies]

closed 2020-03-21 by CTCaer