CTCaer/hekate #275: Ini folder configurations list in Auto Boot configuration is empty

ThatNerdyPikachu, 2019-07-27:
```
bootloader/
    ini/
        0-android.ini
```

Doesn't show up in ``Options -> Auto Boot``

CTCaer, 2019-07-28:
This will might be a bug with my parsing and list preparing for autoboot window in Nyx.
(Because it works for booting).

I'll check it.

Lesking72, 2019-07-29:
The ini folder configurations appear if there are options in hekate_ipl.ini, but they seem to take the name of options that are in hekate_ipl.ini

For example, if I have a boot option named ReiNX in hekate_ipl.ini and an ini file named Lockpick, then the auto boot menu will show "ReiNX" under both main configurations and ini folder configurations, but the one in ini folder configurations is actually Lockpick.

CTCaer, 2019-09-09:
https://github.com/CTCaer/hekate/commit/6159d94e5df232877f7ace9490a814a2fa36fbe7

closed 2019-09-09 by CTCaer