CTCaer/hekate #277: Create EmuMMC on proper partition according to MBR table codes.

Zexceil, 2019-08-06:
When creating EmuMMC in Hekate if you're using a hybrid MBR then the sectors 1-2047 will be marked as EE for the GPT structure stuff; hekate is then trying to write to the next partition in the table which is the general partition used for HOS and etc.

If hekate was able to acknowledge the EE code which would cause it to not register that partition then skip the 1st partition as that's general data(like it does now), it'll allow it to write the EmuMMC in the correct location.

Don't quote me on the code marking and their exact meanings; I don't exactly know a huge amount the codes and the meanings, just the general gist.

fennectech, 2019-08-06:
Its recommended NOTto have this EE code as it prevents hos from booting at all.

CTCaer, 2019-08-07:
Official release has Part 1st: 2048 - end as xEE and Part 2nd: 1 - 2047. 
Also first part must always be the fat32 offset.

So how it tries to write to second part which is only 1MB?

Have in mind, that I intend to protect xEE parts. The thing is that android is very experimental on that side of it.

For example the MBR isn't utilized correctly. It should be sth like that:
Example:
1st Part: fat32 offset and size,
2nd Part emuMMC,
3rd part: PGPT sector 1 offset and size the whole disk
and gpt can have the same for part 1st and part 2nd so they can be manipulated in a linux PC.
Ofc there's a always an easy way of doing things here. PGPT can go to 4th part and never changed after that, because it covers the whole disk. This means that you can do GPT partition manipulations (with an app that doesn't break MBR) or utilize part 2nd/3rd for MBR with ease. And you can put whatever offsets you want.

multiple xEE should not be used. What happens in linux currently, is that it checks 1st part for GPT and it fails. Then sees that there's a 2nd one and it succeeds. hybrid MBR is power and a curse, so it should be correctly used.

CTCaer, 2020-05-01:
New Nyx has a partition manager to ease that process

closed 2020-05-01 by CTCaer