CTCaer/hekate #279: ASK Migrate EmuMMC to NEW SDcard

okaka14, 2019-08-14:
So I have my switch set up with latest Kosmos/SD partition raw **EmuMMC** on fw 8.1.0/ and I would like to migrate to a new bigger-sized MicroSD...For the SD files I know I could just copy them over but what about the EmuMMC? How do I migrate that over to the new MicroSD? Also, anything else I need to watch out for when I migrate? 
Thanks!

fennectech, 2019-08-14:
There is no easy way to do this.  Your going to need to backup the emummc partition to your computer using an disk imaging tool. Then restore it to a partition on the new sdcard.

Weltenfeind666, 2019-08-28:
maybe I got the same problem: how do I copy my old mSDcard data (atmosphere (FW 8.1.0 (AMS 0.9.2.)
from an old mSDcard to a bigger one? If I touch launch it tried to launch and I'm back in the home menu :( what do I have to do? thanks

CTCaer, 2019-08-29:
EmuMMC partition based, can only be imaged with a tool for now.
And then copied back to a new partition.
After that you run migrate and it finds it.

Weltenfeind666, 2019-08-29:
could you please help me how to this? I was asking in GBAtemp but I got an answer like: copy and paste works, and thats not correct. which tool do I use and If I do an image how works it with the different sizes of the mSD-card image (128gb to 400gb) thanks

urherenow, 2019-08-29:
Win32 disk imager. First, put your original SD in the computer. It should say something like “this disk must be formatted before it can be used”. DO NOT FORMAT. If you did, your emummc is already gone. You should see 2 drive letters pop up. 1 should show your files (like hekate and such). The other, will tell you again that it needs to be formatted. This is the drive you want to make an image of. Pick a name/location, and click “read”. 

When that’s done, put your new card in, and select “write”. Copy all of your files to what’s left over (which should be fat32, if not, format it as such).

CTCaer, 2019-08-29:
Btw format will not kill it because I added a 16MB offset. But it's generally the best for that advice (DO NOT FORMAT) to be followed.

As for the order of doing things, it's incorrect.
HOS does not like the first part to not be the FAT32 one.

So it needs to be partitioned first (minitool, disk genie or whatever other free tool).

Make a big partition that you will format it as FAT32 and make sure that at the same window, it says that unallocated space AFTER is as big or bigger than your image. 
You can use the drop downs to select a more fine tunable unit (MB for example).
(Also enable the align partitions to 1MB).

Then you create a another primary partition (emuMMC) that takes what's left. And make sure you choose unformatted for that one. It will create the partition but not format it.

Now you can use a tool to restore the image you took of the emuMMC partition onto the 2nd partition of your new sd.

Weltenfeind666, 2019-08-29:
how do I write it on the (second partition which windows cant handle) from the new sd card (partition) the tool tells me that it will damage my device and after I tried it fails and gives me Error 5

Weltenfeind666, 2019-08-29:
[image]
I've done that also with the bigger mSD-card 

how do I write it on the (second partition which windows cant handle) from the new sd card (partition) the tool tells me that it will damage my device and after I tried it fails and gives me Error 5

Weltenfeind666, 2019-08-29:
I think it's the problem that my image is to big for the unformated partition. can I shorten the emuMMC if I delete some software from my switch and try the same? I followed the guide from the site from which I have the screenshot.

urherenow, 2019-08-30:
Win32 disk imager should be able to write the image to it.

CTCaer, 2019-08-30:
>Make a big partition that you will format it as FAT32 and make sure that at the same window, it says that unallocated space AFTER **is as big or bigger than your image**.
You can use the drop downs to select a more fine tunable unit (MB for example).

I told you to make the emuMMC as big or bigger than your emuMMC partition image size.

Just resize the fat32 one and move and resize the unformatted also. Or do it from start and DON'T follow that bit:
[image]

Check your image size. That or bigger must be the size.

Weltenfeind666, 2019-08-30:
If I do so, I don't have much from the bigger mSDcard - it would be a waste of space or do I understand that wrong?

urherenow, 2019-08-30:
You don’t seem to grasp that you’re “wasting” the very same amount if space on your SMALLER card already...

Dramkissoon, 2019-09-04:
> EmuMMC partition based, can only be imaged with a tool for now.
> And then copied back to a new partition.
> After that you run migrate and it finds it.

After I restore the image to my new bigger micro sd card, I extend the extra unallocated space and in Minitool I can see the partition for emunand that I had in my original micro sd but when I boot up I see that in "Change EMUMMC" it's blank and Migrating does not find it. Any ideas?

urherenow, 2019-09-04:
Did you make your unformatted partition (for restoring your emummc) at the END of the disk? The first partition (after the initial 16MB) MUST be your fat32 partition.

That’s my understanding, anyway.

CTCaer, 2019-09-04:
As I said before, stop mangling the emuMMC backup images or the drive.

Create a backup image of the partition -> create a partition that FITS that image in the new drive and restore it.

Also win32 disk image cannot be used for that reason because it images the whole drive. dd is a better solution

CTCaer, 2019-09-04:
As for the issue that really is a hekate request, this will be implemented in the future.

Dramkissoon, 2019-09-04:
@okaka14 did you manage to migrate your emummc successfully yet? I've been trying this but no luck.

urherenow, 2019-09-04:
Win32 Disk Imager certainly can be used. It’s all I ever use. It images an entire drive letter, not an entire physical drive. If you make the correct size partition, UNFORMATTED, and that partition has a drive letter, an image is written to it no problem.

Doesn’t even have to be unformatted if you know the correct drive letter. Unformatted just makes it the default location that Image Writer picks.

CTCaer, 2019-09-05:
hmm, last I tried it, it did a full disk image. MBR and all.

boongee, 2019-11-28:
For Windows users, I've found [ODIN](https://sourceforge.net/projects/odin-win/) to be an excellent tool for creating backups of the emuMMC partition.  Within Options, select **Disk Options/Save all blocks** prior to creating your backup.

victorgneves, 2019-12-23:
Use Emutool, it's a software that does everything for you.
It even converts EmuMMC to SXOS EmuNAND.

https://gbatemp.net/threads/emutool-move-partition-emu-on-another-sd-switch-emu-type-on-sxos-and-more.550756/

YolloPlays-AI, 2020-02-16:
I just wanted to ask if that is a feature now? I was just wondering about it because it would be really useful.