CTCaer/hekate #281: atmosphere (FW 8.1.0 (AMS 0.9.2.) with new mSDcard

Weltenfeind666, 2019-08-28:
I've copied my old mSDcard data to a new bigger mSDcard but it failed to launch if I choose Atmosphere FSS0 EmuMMC - I get back to the launch/home menu

what is to do? thanks

Weltenfeind666, 2019-08-28:
yes I did, fat32

urherenow, 2019-08-29:
Did you format the card first? They generally come as exfat...

urherenow, 2019-08-29:
Sorry for the delay... I’m on Tokyo time and just finished work. So how exactly did you transfer your emunand? Win32 disk Imager (ImageWriter)? DD on Linux? You can’t simply copy and paste a partition...

Weltenfeind666, 2019-08-29:
no problem, I'm very grateful to get answers and love the good work for the switch ;)

Iust copied my mSDcard data without any image writer, so that seems to be the problem - I use windows 10. but please tell me what to do that I can use my new mSDcard :) thanks!

urherenow, 2019-08-29:
Not to be rude, but github is for reporting bugs, not a help forum. Please move your question to gbatemp and I’m sure you will get plenty of answers.

CTCaer, 2019-08-29:
Nyx has a Logs button before launching a boot entry.
Enabling it, will show boot log.

See what step fails.

Example:
[image]

Weltenfeind666, 2019-08-29:
I touched logs off to logs on, but if I touch launch Atmosphere FSS0 EmuMMC there is only the screen with HEKATE - IPL Custom NX Bootloader

CTCaer, 2019-08-29:
Ok then do the following:

Rename /bootloader/sys/nyx.bin to nyx.bin.bak
Launch hekate and use the volume buttons and PWR to navigate the text UI.

Choose `Atmosphere FSS0 EmuMMC` and tell me what happens.

Dramkissoon, 2019-09-04:
> Ok then do the following:
> 
> Rename /bootloader/sys/nyx.bin to nyx.bin.bak
> Launch hekate and use the volume buttons and PWR to navigate the text UI.
> 
> Choose `Atmosphere FSS0 EmuMMC` and tell me what happens.

Im having the same issue where it just brings me back to hekate launch screen after trying to boot cfw but for a brief second after the kosmos splash screen, it says "Unknown pkg1 version" in red.

CTCaer, 2019-09-04:
Not copying the emuMMC correctly or restoring wrong BOOT0/pk2 combinations are not an issue with hekate. For anyone that for some reason does not find a solution, closed issues here were answered repetitively with most possible causes and fixes.

closed 2019-09-04 by CTCaer