CTCaer/hekate #289: not working pkg2 decryption failed

seawave11, 2019-09-15:
there something wrong with my nand or the keys for 9/0 updated fully yet?

seawave11, 2019-09-15:
then i tried to dump the keys it says unknown

CTCaer, 2019-09-15:
You might want to included more info:
HOS version that you are trying to boot?
Sys or Emu?
hekate 5.0.2?

seawave11, 2019-09-15:
hekate ctcar 5.02
is system not emu
hekate 5.02

CTCaer, 2019-09-15:
You sure you inject hekate 5.0.2?
That error can only happen on 5.0.1 and lower

seawave11, 2019-09-15:
by inject you mean into firmware? no i don't think so,
i'm on newly firmware so the files on the sdcard old?

CTCaer, 2019-09-16:
How you run hekate?

seawave11, 2019-09-16:
it was injected by usb and yes it hekate_ctcaer_5.0.2

CTCaer, 2019-09-16:
Show me your hekate_ipl.ini

seawave11, 2019-09-16:
[config]
autoboot=0
autoboot_list=0
bootwait=1
verification=1
backlight=100
autohosoff=0
autonogc=1

{AtlasNX/Kosmos v13.2}
{}
{Discord: discord.teamatlasnx.com}
{Github: git.teamatlasnx.com}
{Patreon: patreon.teamatlasnx.com}
{Pegascape DNS: pegascape.sdsetup.com}
{}

{--- Custom Firmware ---}
[CFW (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
{}

[CFW (EMUMMC)]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
{}

{--- Stock ---}
[Stock (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1
icon=bootloader/res/icon_switch.bmp
{}

CTCaer, 2019-09-16:
Dunno what you do but you don't run hekate_ctcaer_5.0.2 but an older version.
Or you have something strange at /sept/payload.bin that is a fake 5.0.2 version.
(hekate does a version check there)

seawave11, 2019-09-16:
so that size is of the payload happen be 5.193 mb

CTCaer, 2019-09-16:
Go to hekate menu and take a photo of it

seawave11, 2019-09-16:
how can i do that from hekate ipl menu?

CTCaer, 2019-09-16:
Use your phone

seawave11, 2019-09-16:
[image]
[image]
[image]

CTCaer, 2019-09-16:
Well I asked for the main menu. The one that shows Launch/Morecfg/Payloads/emummc.

Your Nyx is old for sure. You are supposed to drag n drop the bootloader folder and replace the files.

seawave11, 2019-09-16:
ok my bad

[image]

MACandCHEEZWIZ1, 2019-09-16:
Not that I'm trying to hijack, but I'm having a similar problem to this and also this
https://github.com/CTCaer/hekate/issues/288

In short,
-I'm using 5.0.2
-I formatted my SDXC card to FAT32 according to the guide on NH Switch and downloaded all the latest updates to Atmosphere, Hekate, etc.
-I'm currently on OFW 9.0
-I backed up BOOT0/1, RAW GPP, SYS, and USER. For Dump package1/2, I get "Found pkg1" but "Pkg2 decryption failed!"
-It refuses to let me launch FSS0 EmuMMC, FSS0 SYS, or Stock SYS. Every time I choose one, it cycles back to the Hekate IPL boot up screen and then the main menu. I'm still able to boot the console normally.
-I did try using the ReiNX payload not long before this, but the splash screen was really quick and didn't load into Atmosphere at all.

CTCaer, 2019-09-16:
The issue is that all 3 of you never updated sept.
Sept changed to add 9.0.0 support. Without it, you get that pkg2 decryption error.
The unk pkg1 in nyx is simply because old nyx.

Basically update your files and stop leaving out the important ones when nitpicking or not.

seawave11, 2019-09-16:
so that why i get that pkg2 decryption error as booting ofw haves no problems

m4xw, 2019-09-16:
I dunno why we do all the effort to create drag & drop bundles only for users to not drag and drop them

MACandCHEEZWIZ1, 2019-09-16:
> The issue is that all 3 of you never updated sept.
> Sept changed to add 9.0.0 support. Without it, you get that pkg2 decryption error.
> The unk pkg1 in nyx is simply because old nyx.
> 
> Basically update your files and stop leaving out the important ones when nitpicking or not.

Aaaah, I see. I must have switched the sept folder over to an older version when I downloaded ReiNX 2.4. Everything was working before I added all the ReiNX stuff.

seawave11, 2019-09-16:
I really don't know if you know idea either cuz I've tried with you said it still doesn't work it the firmware updates did someting

CTCaer, 2019-09-17:
Update your CFW.
hekate is only the bootloader for it.
Download the latest version of your CFW and drag n drop the files.
Especially the **sept** folder.

Same goes with hekate. Download the latest release (hekate v5.0.2 & Nyx 0.8.2) and drag n drop the bootloader folder to your sd.

This issue is not a hekate issue but old files that do not support 9.0.0.

closed 2019-09-17 by CTCaer

mrvnklm, 2021-01-17:
@seawave11 Did you actuall find a solution? I am facing exactly the same issue...

MACandCHEEZWIZ1, 2021-01-18:
I think I just did a clean install in the end. I kind of hate updates
because updating one thing can make everything else stop functioning. One
of the annoying things was that the SDsetup file downloader page had a
Trojan in one of its files.

In short, homebrew is a huge pain.

urherenow, 2021-01-18:
No, taking shortcuts is a pain. Do it right, and grab the latest from the source, and you won't have any issues. Stay away from "packs".