CTCaer/hekate #290: sd and emmc data <@LBA 00100000>, do not match

Tensepanda, 2019-09-18:
I get, "sd and emmc data <@LBA 00100000>, do not match", when it goes into the green bar when I try to make a backup nand. 

Just bought the card from a retailer, Sweden. 

Am I doing something wrong, or do I need to buy a new card? 

Best Regards

TensePanda

CTCaer, 2019-09-19:
Can you go to `Console info -> Sd card`, then tap the screen with 3 fingers and upload the screenshot here?
(Alternatively just take a photo of that screen)

Tensepanda, 2019-09-19:
Hi. 
Thanks for the reply, I sorted it out. 
Instead of holding down the Volume down button to boot into the menu. 
I just booted directly into Hekate, from there I made my backups, no problem at all :) 

Sorry for taking your time, and I want to say thank you for all the time you put into this. 
Much appreciated, lots of love from Sweden. 

Best Regards

TensePanda

closed 2019-09-19 by Tensepanda

CTCaer, 2019-09-19:
@Tensepanda What do you mean you booted directly into hekate?

Tensepanda, 2019-09-19:
Bare with me, I'm kinda new to this.

If I just power up Hekate with a loader, I get like a menu with visual buttons etc.
On the top you get things like options, tools and what not.
On the bottom I think you have something like reload, reboot and power off.

When I got the error I tried saving my backup through the other menu, the one with only texts and no images.

Launch
Option
Tools
etc
etc
etc

But I later tried backing up my Nand through menu with images and whatnot. :)
Sorry if my explanation is kinda vague :/

Best Regards

TensePanda

[https://ipmcdn.avast.com/images/icons/icon-envelope-tick-green-avg-v1.png]<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>     Virus-free. www.avg.com<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>

CTCaer, 2019-09-19:
Hmm, I expected the opposite.

Btw, the GUI (graphical UI) is called Nyx. The other one (hekate main), you can call it TUI (text UI).

Why pressing VOL- didn't launch Nyx?
Did you use an old hekate version (latest is v5.0.2)?
Nyx was introduced on v5.0.0.

Tensepanda, 2019-09-21:
I used 5.0.2, well holding volume - did boot it, but only sometimes, I have 2 switches, both unpatched.

But everytime I tried to do a nandbackup it failed half way through the green bar.

So I just did a nandbackup through the "normal" nice looking menu :)