CTCaer/hekate #291: Boot to RCM payload fail most of ther times on hekate_ctcaer_5.0.2, boots fine(instant) on 5.0.1

Ntemis, 2019-09-22:
Hi, i have an RCMloader zero and boots instantly when used with 5.0.1 payload.
When i updated to 5.0.2 i though my dongle broke :)
In cold boot no way, it lights sometimes the led and then switches of, i have to go to official os then shutdown the retry and if lucky my sysnand is 8.1.0 not sure if that's the issue here

CTCaer, 2019-09-22:
What happens if you put 5.0.2 to /bootloader/update.bin and use 5.0.1?

Hmm that's weird anyway. So a shutdown works sometimes, but injecting never?
I may need to send you a couple of payloads to test btw.

Ntemis, 2019-09-22:
Black screen backlight off, and ofc i will test what you suggested and report back but also am around for anything need to be tested.
Thanks for the support

Ntemis, 2019-09-22:
ok i tested "put 5.0.2 to /bootloader/update.bin and use 5.0.1"
cold boot no go, boot to ofw and reboot + vol+ ok

CTCaer, 2019-09-23:
@Ntemis Do you have discord?
(You can find me in retronx one)

Ntemis, 2019-09-23:
Issue is fixed thanks to @CTCaer closing down

closed 2019-09-23 by Ntemis

CTCaer, 2019-09-23:
Issue was confirmed to be a miss pushed change of force shutdown timeout.
https://github.com/CTCaer/hekate/blob/master/bootloader/soc/hw_init.c#L223

CTCaer, 2019-09-23:
Restored back via https://github.com/CTCaer/hekate/commit/2857fcca529e1419aa1be2b054a3dc5f4e68fd2f

Motz0207, 2019-10-10:
need help
i just accidentally updated my firmware to 9.0.1 and i already downloaded the files hekate_ctcaer_5.0.2 but i dont know how to update. can anybody help me?

Motz0207, 2019-10-10:
> Issue was confirmed to be a miss pushed change of force shutdown timeout.
> https://github.com/CTCaer/hekate/blob/master/bootloader/soc/hw_init.c#L223

can you pls help me? i accidentally updated my nintendo switch to 9.0.1
i already downloaded the hekate_ctcaer_5.0.2_Nyx_0.8.2 but i dont know what to do. i dont know hot to update the system pls help