CTCaer/hekate #292: Using Hekate to launch stock

KaNawogirusa, 2019-09-27:
Hi, I just want to use hekate to launch my switch in Stock so I don't have to remove autoRCM, what are the minimum required files that I have to put in the SD that I'm going to use online with original games?
Thanks!
PS: I have emuemmc on my other SD so I think it's reasonably secure to use online in stock mode right?

CTCaer, 2019-09-28:
Files:
```
/atmosphere/fusee-mtc.bin          (used by sept. 7.0.0+)
/atmosphere/fusee-secondary.bin    (used by hekate to pull exosphere/warbmoot. 7.0.0+)

/bootloader/hekate_ipl.ini         (hekate cfg)
/bootloader/sys/libsys_lp0.bso     (used by hekate for sleep in HOS)
/bootloader/sys/libsys_minerva.bso (used by hekate to speed up boot)

/sept/sept-primary.bin             (used for generating keys. 7.0.0+)
/sept/sept-secondary_00.enc        (used for generating keys. 7.0.0+)
/sept/sept-secondary_01.enc        (used for generating keys. 7.0.0+)
```

hekate_ipl.ini:
```ini
[config]
autoboot=1
autoboot_list=0
bootwait=0
verification=1
backlight=100
autohosoff=0
autonogc=0

[Stock]
fss0=atmosphere/fusee-secondary.bin
stock=1
emummc_force_disable=1

```
`autoboot=1` will automatically boot [Stock] on injection
`bootwait=0` will disable bootlogo and start booting immediately. If you want an indication but still be fast, change it to 1 second.
`emummc_force_disable=1` in the ini is there just in case you have emummc enabled in the **same** SD card.

These are exactly what you need to boot stock and only stock on 7.0.0+.

KaNawogirusa, 2019-10-01:
Worked like a charm, I hope nintendo doesn't ban me now haha, thanks a lot !

closed 2019-10-14 by CTCaer

UBFI, 2021-01-19:
Is it safe to boot with the fusee payload and not risk a ban?

impeeza, 2021-01-19:
UBFI: A Issue for support is no the best place, please read the documentation, and all the great guides on the net.