CTCaer/hekate #293: Hekate frozen on bootlogo

DoonerDX, 2019-09-29:
I may be missing something. But when I send the Hekate payload. The bootlogo will show up but the switch will never actually run Hekate. I was attempting a downgrade through choidujour. Thinking that I had a problem with the boot0/boot1. However I need to run Hekate to complete the process. Any ideas would be welcome. And maybe I'm just being a noob and a patched switch will display a bootlogo but not go any further. I can run lockpick and biskeydump. Thanks.

NickHam13, 2019-09-29:
You're not on a patched Switch, they can't launch payloads through RCM.

DoonerDX, 2019-09-29:
Thanks for the confirmation NickHam13. I've tried pushing different payloads and all of them seem to hang and not load anything past their bootlogo. Have also tried older versions.

CTCaer, 2019-09-29:
@DoonerDX can you delete /bootloader/sys/nyx.bin and tell me what happens?

DoonerDX, 2019-09-29:
@CTCaer deleted. When payload is sent screen is left blank.

CTCaer, 2019-09-29:
delete the whole sys folder. and if boots to the menu, go to console info -> sd card info

DoonerDX, 2019-09-29:
Blank screen again. No bootlogo or anything.

CTCaer, 2019-09-29:
Can you try hekate v4.10.1 ?

DoonerDX, 2019-09-29:
Tried the Hekate v4.10.1 on its own from the releases. It only showed a blank screen. Also got the kosmos with Hekate v4.10.1. It brought up the blue atlasNX kosmos screen then went black with a small blue strip on the left hand side.

CTCaer, 2019-09-29:
oh I see. sd card issues.

Boot without sd card. Then plug it in and check the console info -> sd card info.
(doesn't matter which version of hekate)

DoonerDX, 2019-09-29:
Sorry still nothing. Black screen. I assume there must me some serious system error. I will try and rebuild the system using choidujour and see how that goes. Thanks for your help.

CTCaer, 2019-12-08:
This was fixed in latest release https://github.com/CTCaer/hekate/releases/latest

closed 2019-12-08 by CTCaer