CTCaer/hekate #294: Really needing help here

BigBossVII, 2019-10-07:
Homebrew isn't loading up like it usually does through the photo album. my firmware was 8.0.1 and I updated it because I wanted to install a theme. I used ChoiDujour to update it to the lastest firmware which was 9.0.1.. Didn't work so well (go figures, I'm fairly new to this so that was a mistake). Now I'm simply trying downgrade but I can't get to the Homebrew page. Any help with the right direction would be greatly appreciated!!!! 

![thumbnail] (https://user-images.githubusercontent.com/56238747/66278203-b9a59e00-e874-11e9-9226-96f6afe6eb4a.jpg)

NickHam13, 2019-10-07:
Hold R and launch a game to enter hbmenu.

fennectech, 2019-10-07:
your gonna wanna update atmosphere on your sdcard   you likely have an old version installed

CTCaer, 2019-10-07:
Update atmosphere files and sept.

That message in 9.0.x shows up because of old sept.
But even then if you don't update atmosphere, you will hang.

hekate is only a bootloader. It takes the atmosphere files or whatever you tell it and tries to boot them. If these files do not support latest hos, even if hekate does, you will not boot.

BigBossVII, 2019-10-07:
you have saved a life !_! Thank you so much, it’s fixed now

closed 2019-10-14 by CTCaer