CTCaer/hekate #296: Hekate 5.0.2, Atmosphere 9.0.4, EmuMMC boots to Hekate

YetAnotherOrc, 2019-10-08:
[image]
New to Nintendo HB (sony pro) and I everyone said to update from 7.0.1.
I have my stock SD card which launches Atmosphere(forget if I hit emuNand or MMC, and don't know if I'm already marked for life from that..)
I have a larger SD which formatted via MiniToolPartition like the guides all say, tried the 16mb lead and both sides to Fat32 which always puts #2 to unspecified but still goes through the emuMMc formatter without error, copy directly from my working card  apply it to the emuMMC slot and launch via emuMMC
So I'm just hoping either emuMMC gets an update for it, and that since I never launched homebrew within Atmosphere(other than checking the album confirming HB was available), that I'm not already banned for life and can one day launch from within the safety net. I did do Nand Backup prior soo now I'm just waiting to tryout the Sky Dungeon addon for my retail BotW
Oh and grinding my teeth if someone could direct me to the best place to post this if this is not the best pleases thank yous!
I think I added a photo of my partitions but, first day on here for exactly this posting.

CTCaer, 2019-10-08:
What's the error that hekate shows briefly in red text?
You can also try to boot after turning Logs ON from Launch menu.
Also who said that 7.0.1 is recommended and why?

(On another note, your emuMMC partition is too big and wastes space. The minimum size that it should be is 29844 MiB. You are 33.35 GiB above that target.)

YetAnotherOrc, 2019-10-08:
Crazy thing is, there is never a message. Ever, slow mo video, burst pic no red, just splash screens.
I think it was one of the videos that said it was best to update to 9.0.1 for compatibility with newer stuff
(Assumed the emuCard was going to host whatever I saved during HB and the videos didn't clarify while one went over from the same author)

I've tried the various partition setups with files from the preconfig zipper site, and also with lastest GitHub releases, never getting any red error message visible to the naked eye nor digital. But, the same files Will boot atmosphere from my original backup SD card too small for both partition and NAND backup (also Backed to PC and not sure if my system is just compromised anyway, since I did launch Atmosphere into the Rphoto for thinking I had my other card inserted)

YetAnotherOrc, 2019-10-08:
On the Logs side, I flip it on every time and it still shows nothing extra, but my More Configurations tab says no .ini or boot entries found... check bootloader/ini/ (which doesn't exist at all from my photo..) for at least one entry

I guess I'll try a new folder named ini and put in atmosphere LOADER? or sept primary..
Or should I wait until someone who's done this before keeps me from bricking my stuff

CTCaer, 2019-10-08:
How you created the emuMMC?
Have in mind that emummc.ini is unique for each card. That's because the offsets of sectors are different.

And I just saw that I missed a case of error. That's when I check if emummc ini succeeded in read PKG2 function. If failed, it returns immediately to hekate.
So it seems the issue is that you copy the whole old sd card, together with emummc.

Go to emuMMC -> Migrate. This will fix your offsets.

YetAnotherOrc, 2019-10-09:
Primary SD never had partition
Big SD MiniTool Partition Wizard 11 from Tuts. emuMMC created from partition, via Hekate AFTER data migration. 
Deleted contents and added the stock bundles before Hekate emuMMC recreation (since emuMMC is deleted), but it's still the exact same nothing, Launch, AtlasNX Cosmos All-in-One splash, Hekate Splash no errors.
I Think it would not have a file in the way since there was no partition on the first card. If there was no conflict I believe that it would have run the process during Hekate emuMMC Management -> Create. Same page, emuMMC info + select: SD Raw Partition... It was made the only way the prescribed partition tool would allow during format, but I feel this may be the problem. Still very suspicious that the missing /bootloader/ini/ (Folder) has no .ini or boot entry. Loads of .ini in the bootloader folder btw.
So, Bad RAW partition? Nonexistent folder? While this is pondered, I'm going to try Just the latest release individual downloads on top of the Fat32(which on apply results in mysterious unformatted partition) and just cross my fingers. Gulp

YetAnotherOrc, 2019-10-09:
Update -- Found the GitHub release Does have the ini folder.. repartitioned and about to create emuMMC once the copy is done

YetAnotherOrc, 2019-10-09:
Soo Wooow this is not the same folders or UI as the last thing. My bad for assuming everything with the same name was on the same page.. Definitely helps to have another mind to bounce off of at my stage..
Either way Im about to attempt figuring out Launch message: no payloads or modules found. Thank you for your patience so far, I'm sure I'm going to be wrong about the rest of the process but I'll take a stab at it, but more assiste would be appreciated.

Odd though that the advanced UI was not in the same version of the same title. I'm losing my mind most likely.
payloads/modules.. insert atmosphere folder? and contents same location, to cover bases.. My console is doomed lol, I swear I'm smarter than this usually "P

CTCaer, 2019-10-09:
First of all, please try to make structured messages. Blurting wall of texts with everything that comes to mind, will not help me understand easily what you are doing wrong.

So lets start:
1. More cfg, aka /bootloader/ini stuff, are only needed if you use them.
2. Launch saying no boot cfgs, means that your hekate_ipl.ini does not have a boot entry now
3. I don't understand when you are saying advanced UI. Before you said that you used the emuMMC management. This falls in the hekate GUI called nyx. So what do you see differently?
I bet you never copied the bootloader folder from the release and the "advanced" ui you see is actually the fallback text UI. Copy the bootloader folder from the release.

4. Focus on your actual problem and don't ask 1 million irrelevant questions or go into panic mode for some reason.
Otherwise I'll close the issue.
Issues page here is for reporting actual bugs or requesting features, etc. Not for support.
I still do some support here, but I need proper troubleshooting from your side and proper reporting of your issues. I ain't gonna troubleshoot your every-minute-changing setup.

So, as I said before, the only error that is not displayed is the `emuMMC init failed` error. So make sure your emuMMC partition is fine and that the emummc.ini file is correct.
Again, your issue here is that you are mixing emummc stuff and breaking configurations after the emuMMC creation from hekate.

iharosi, 2019-10-10:
```
Initializing...
Found pkg1 (’’).
Unknown pkg1 version.
```
That's what I can see when trying to boot into emuMMC with Logging turned on then it immediately falls back to nyx.
Unfortunately my console has been updated by its pervious owner to 9.0.1.
I'm not sure if this version is supported at the moment by [emuMMC](https://github.com/m4xw/emuMMC) as their GitHub page says:
_„Supported Horizon Versions
1.0.0 - 9.0.0”_

This is my hekate_ipl.ini:
```
[config]
autoboot=0
autoboot_list=0
bootwait=3
verification=1
backlight=100
autohosoff=0
autonogc=1

[Atmosphere with emuMMC]
fss0=atmosphere/fusee-secondary.bin
logopath=bootloader/bootlogo.bmp
atmosphere=1
```

I did not use anything but only [Atmosphère 0.9.4](https://github.com/Atmosphere-NX/Atmosphere/releases/tag/0.9.4) and [hekate - CTCaer mod v5.0.2 & Nyx v0.8.2](https://github.com/CTCaer/hekate/releases/tag/v5.0.2) on the SD card. And partitioned as most of the guide suggest. The emuMMC partition has been successfully created by Nyx.

I understand this is not a support page but would be nice to know the above described issue is related to the lack of support of 9.0.1 or am I doing something wrong. Thanks.

CTCaer, 2019-10-10:
Press the migrate button in emuMMC and tell me what it said. (it's OK to press continue)

iharosi, 2019-10-10:
`No emuMMC or foreign emunand found!`
That's interesting as the emuMMC has already been enabled. Also, the creation of emuMMC has been succeeded as well. What did I do wrong? :S

iharosi, 2019-10-10:
Okay, here is what I've found and very interesting. I've deleted the `emuMMC` folder, recreated and enabled emuMMC with Nyx. After this the `Migrate emuMMC` option said it found the partition and can fix the config. I moved forward with it pressing continue. I've checked once again this option and it said the same, so everything looked okay. Once I tried to boot CFW emuMMC, I've got the same red message (`Unknown pkg1 version.`) then I got back to Nyx, then checked `Migrate emuMMC` option once again and it said no emuMMC found. Now I've got lost. o.O

CTCaer, 2019-10-10:
Heh,  it actually makes sense and it's bad news for you.
Unfortunately you have a fake sd card.

iharosi, 2019-10-11:
Totally makes sense!
I've bought a new SD card yesterday and everything went smoothly without any issues, so definitely it was a fake SD card.
Thanks for the tip!

CTCaer, 2019-10-11:
@iharosi good to hear
@YetAnotherOrc what about you? You did the steps in order to pinpoint the issue?

YetAnotherOrc, 2019-10-14:
Firstly, thank you/my bad  again for not submitting this on the public forums,
secondly my bad for having such a cluttered downloads folder that can't make it easy to distinguish which origins are for which issues
My issue was first from a hekate 5.0.2 bin injection into the AtlasNX which had the emuMMC options. Without the emuMMC it would work on my old card And my new card. Add emuMMC, deLaunch no messages.
In my cluttered mind I started clean and used your release and bin injection and yes it was a different interface with volume/power buttons for control, missing assets for sleep and other, and Launching led to to black screen of indeterminate length (aborted by extensive Power button holding TYJeebus)

So after some overtime at work, visiting my incapacitated loved ones, and cleaning the crap out of my home and computer, I managed to find an updated Atlas release (works without partition) and fed through a brand new emuMMC partition.

"Unknown pkg1 version" in the farthest corner sideways, impossible to read speed even with slowmo without finger calibration.
------------------------
So it looks like I also have the bad sd card which was suspected.
------------------------
Gameplan:
My first SD (Nand backup) shall be operated on. 64 gigs, probably more than I need.

If nothing else, I can plan on never playing online again. Fingers crossed I can get my DNS working so I can leave FTP behind. Its a nightmare with Sony and I plan to stop stressing any day now "P

Au Revoir and all that Jazz

YetAnotherOrc, 2019-10-14:
Oops sorry about that Headline text, unexpected result with the publishing interpreter <(^^)>

CTCaer, 2019-10-14:
backup your data and find how big is your sd before proceeding to just use it. (h2testw can be used or any other sd validator)

closed 2019-10-14 by CTCaer