CTCaer/hekate #298: Failed to init SD card. Make sure that it is inserted. Or that SD reader is properly seated!

Samii571, 2019-10-19:
When i try to launch something or when i want to use anything sdcard related, this error message comes out. 
But when i press the sdcard reader (from the case) its working...
Do someone has a solution for this problem?

maskmark, 2019-10-20:
apparently it's a hardware related issue. try make your card reader mini-board steady fixed onto the MB slot.

Samii571, 2019-10-20:
(I come from germany, and even with a translator i didn't understand you properly)So you think I should unscrew the switch and attach the card reader better?
And if so, where should I fasten the card reader, or should I leave it where it is and then fasten it?

maskmark, 2019-10-20:
[image]
well, re-inatall this thing firmly on your main PCB.

maskmark, 2019-10-20:
re-inatall=re-install

ELY3M, 2019-11-20:
I seen few people who have missing pins from the sd reader connector part on motherboard. make sure to check the connector for any missing pins.   use phone camera and zoom.

CTCaer, 2019-12-08:
Closing as this is a hardware issue

closed 2019-12-08 by CTCaer