CTCaer/hekate #303: Blackscreen after injection with hekate >= 5.0.0

Mega349, 2019-10-29:
I tried three different Nintendo Switch´s but one wont work (same Setup).
If i try inject a Hekate Payload >= 5.0.0 i just get a Blackscreen...
4.* before worked without problems but >= 5.0.0 just without SD-Card.
I can run other payloads without problems.

Sorry for my bad English but if you need more informations just write me on Discord (Mega349#4434)

if someone have a fix or a solution it would be nice if you write me too :)

CTCaer, 2019-10-29:
find me in retronx discord

CTCaer, 2019-11-11:
Issue is validated and it's because of SoC binning.
Running SCLK at 588MHz instead of 608MHz fixes that.
Next version will mitigate this and will still keep performance to high levels for Nyx.
Best case scenario, through some software optimizations, the perf will be almost the same or even better.

CTCaer, 2019-12-08:
https://github.com/CTCaer/hekate/commit/0b45a5a11a95a702ecbbfc27f449f2b80c6c8c55

closed 2019-12-08 by CTCaer