CTCaer/hekate #305: Why are Linux partitions skipped?

FichteFoll, 2019-11-06:
I had a small journey into the net as I was trying to figure out why the partition I created with gparted and did not format wasn't recognized.
Turns out the partition had the linux ftype of 0x83 set, which is explicitly exempted in the code and I only thought to check that after reading the source. I then changed the type with `fdisk` to 0x82 and it worked fine.

https://github.com/CTCaer/hekate/blob/74452074f654c25e0de3904ab828119241b36d60/nyx/nyx_gui/frontend/gui_emummc_tools.c#L182

So, why is this the case?

CTCaer, 2019-11-07:
It does what it says on the tin.
Until further warning is added in the creation process, it just protects you from overwriting L4T/linux partitions.

closed 2019-11-09 by CTCaer

FichteFoll, 2019-11-09:
But why make a special case for Linux and none of the others? I was mostly curious but also because of the TODO.

CTCaer, 2019-11-09:
@FichteFoll Because we have linux in switch. Next version will also check for xEE partitions also.

But the change will be that it will ask and make sure that you want to nuke the linux partition.