CTCaer/hekate #308: Potential issue Emmumc migration?

yuukiw, 2019-11-14:
From the sound of it the migration option allows to port one emunand solution to another, is that assumption right?

I tried to migrate from sx os SD file emunand to atmosphere, Hekate found sxos files and copied and renamed the files arcordingly but everytime I want to boot atmosphere with emunand it shows the Sept logo and then goes to a black screen.

Do you have any idea what may goes wrong?

CTCaer, 2019-11-14:
After sept, you need to wait because currently `SD file` based takes 4-8min to boot.
This will be fixed in a later version of emuMMC (it will reach speeds about the same as `SD partition` - raw).
For its development check https://github.com/m4xw/emuMMC

closed 2019-11-14 by CTCaer

yuukiw, 2019-11-14:
I see, thanks for the answer

yuukiw, 2019-11-14:
waited around 15 minutes, still black screen.