CTCaer/hekate #310: RCM Loader ONE can't inject 5.0.2

sztomi, 2019-11-16:
I thought I did something incorrectly, but apparently many people are experiencing this. 

  * [RCM Loader one](https://www.xkit.xyz/) can't inject the 5.0.2 payload
  * 5.0.1 and other payloads can be injected fine
  * Other methods of injecting the same 5.0.2 work fine (e.g. PC)

See this thread: https://gbatemp.net/threads/cant-inject-hekate-5-0-2-using-the-rcmloader-dongle.548632/ 

I don't know if this is an issue in this Hekate build or in the loader itself. I tried contacting the vendor, but there was no answer after two weeks, so it's unlikely they will respond. Is there anything that can be done in Hekate to make it work? I don't know how to troubleshoot/debug this, but I'm happy to gather any additional info if possible.

CTCaer, 2019-11-16:
In v5.0.2 I passed a debug change by mistake in release which was fixed in master:
https://github.com/CTCaer/hekate/commit/2857fcca529e1419aa1be2b054a3dc5f4e68fd2f

Basically what happens, is that you press the power button for more than 2s and the console powers off.
Power button does not need to be pressed for 2s. You only need to tap it.
So basically you **hold** VOL+ and **press once** power.

sztomi, 2019-11-26:
@CTCaer thanks! I can confirm that it works as expected (I never knew you are not supposed to hold power). Since this is already fixed on master, I'll close the issue.

CTCaer, 2019-11-26:
I'll reopen this for visibility until I push a new release.

CTCaer, 2019-12-08:
https://github.com/CTCaer/hekate/commit/2857fcca529e1419aa1be2b054a3dc5f4e68fd2f

closed 2019-12-08 by CTCaer