CTCaer/hekate #314: Update 9.1.0 endless loop

jcmejia, 2019-12-05:
Apparently the recent update 9.1.0 patched something now hekate refuses to properly make a emummc and a backup as well trying to boot emummc cfw Results In a boot loop were it send it back to the hekate menu

james-d-elliott, 2019-12-06:
Until atmosphere's sept is updated (and probably hekate) expect issues. Updating your switch before confirming compatibility with the CFW/exploits you're using should be considered experimental.

CTCaer, 2019-12-07:
You can always make an emuMMC but you can't boot it.
Backup is irrelevant because like creating emuMMC, hekate doesn't even check the version.

Booting 9.1.0 emuMMC ofc will boot you back in Nyx. It's like 9.1.0 sysnand.

Also the flashing message is `Unknown pkg1 version`. (Btw the flashing will be fixed in next version. You will be able to clearly see the error message.)

CTCaer, 2019-12-07:
Also another thing. This is not a boot loop.
A bootloop/loop needs no user interaction.

This is just hekate flashing the error and instead of waiting it goes directly to menu.

0opsie, 2019-12-07:
Hey I was just about to install my first CFW on my switch and accidentally updated to OFW Version 9.1.0.
Is it still safe to continue with the installation or do you recommend waiting till a 9.1.x supporting hekate version is online?

james-d-elliott, 2019-12-07:
If you update, you will not be able to run CFW until a new release is out. First AMS has to be updated, see here: https://github.com/Atmosphere-NX/Atmosphere/issues/706#issuecomment-562496846

CTCaer, 2019-12-07:
You can only do the initial stuff for now.
Like nand backup (and if you want you can create an emuMMC).

Other than that, as james said above, without final support for 9.1.0 (for now) in hekate AND atmosphere, you can't boot CFW.

Steve0the0cube0man, 2019-12-08:
ATMOSPHERE JUST RELEASED A HOUR AGO WHERE IS HEKATE?????? 
WRYYYYYYYYYYYYYYYYYY

jcmejia, 2019-12-08:
Okay so atmosphere 0.10.0 got released but hekate does the same exact thing
that means its gonna need a update.

Steve0the0cube0man, 2019-12-08:
> Okay so atmosphere 0.10.0 got released but hekate does the same exact thing that means its gonna need a update.
> […](#)

so it's gonna take another 3 days or a week?

CTCaer, 2019-12-08:
Support for **9.1.0** was added

closed 2019-12-08 by CTCaer