CTCaer/hekate #319: Latest version doesn't boot

MegaSuperab, 2019-12-08:
After updating to the latest version, hekate straight up doesn't load. The screen doesn't even turn on. I've tested other payloads and they boot up, so it's not my Switch that's the problem.

CTCaer, 2019-12-08:
Did you drag n drop the bootloader folder?

MegaSuperab, 2019-12-08:
> Did you drag n drop the bootloader folder?

I have, and it still doesn't work.

CTCaer, 2019-12-08:
Are you able to join retronx discord so I can provide test builds?

MegaSuperab, 2019-12-08:
Oh. When you said "drag n drop", I somehow didn't process the fact that I had to do so into my sd card. It works now, sorry.

CTCaer, 2019-12-08:
You are welcome.
It is still a bug I suppose.
It wasn't supposed to hang with old minerva. I'll reopen it.

CTCaer, 2019-12-10:
https://github.com/CTCaer/hekate/commit/e4f7928513ebe00c482bb5a3a3b81fc935738471

Will close the issue when hotfix version is out.

closed 2019-12-12 by CTCaer