CTCaer/hekate #320: Can't boot enter NYX on Hekate 5.1.0

CompSciOrBust, 2019-12-08:
On Hekate 5.1.0 I can't seem to get NYX to load when the boot delay is set to zero. I hold volume down but it just loads my boot config instead of going to the menu. If I set the boot delay to three and then hold volume down it works.
I'm using fss0 and sept if that matters (console fw is 9.0.1). Hekate is being loaded from the update.bin file. The initial payload is Hekate 4.6 being loaded from a Rebug SwitchMe chip.

CTCaer, 2019-12-08:
Thanks for reporting it. Another user also reported that.
Please change your config to 1s for now. (`bootwait=1`).

It's a bug in button code where it returns no button pressed when bootwait is 0.

CompSciOrBust, 2019-12-08:
Np. Thanks for the info.

CTCaer, 2019-12-10:
https://github.com/CTCaer/hekate/commit/ae283aef59d853632a7e609c754b22c114a799c1

Will close issue when hotfix version is out

CTCaer, 2019-12-12:
+this https://github.com/CTCaer/hekate/commit/8ff01301cb39f07f3711cb4778e049310257ec28

closed 2019-12-12 by CTCaer