CTCaer/hekate #321: Unable to boot CFW (SysNAND or EMUMMC)

xgorinx, 2019-12-08:
When trying to boot CFW, receiving errors:

Failed to apply 'nosigchk'!
Failed to launch HOS!

Stock and payload fusee-primary do launch, however.

hekate: v5.1.0
System: 9.1.0
Atmosphere: 0.10.0

Goma1337, 2019-12-08:
Came to report the same error. 

9.0.1 EmuMMC boots fine, 9.1.0 SysNAND doesn't.

Not sure if it's a problem with applying the patches or with the patches themselves. Using Joonie86's patches (https://github.com/Joonie86/hekate/releases).

CTCaer, 2019-12-08:
Failed to apply a patch, means the patch is missing or empty.

Also no piracy here

closed 2019-12-08 by CTCaer

xgorinx, 2019-12-09:
I'm sorry to ask, but what does that mean exactly?  I haven't pirated anything.  I use atmosphere so I can use Edizon.

urherenow, 2019-12-09:
Then you don’t need to enable sigpatches. Those are for installing games and such.

xgorinx, 2019-12-11:
I found it.  In the bootloader folder, the hekate_ipl.ini config file said 

{--- Custom Firmware ---}
[CFW (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
**kip1patch=nosigchk**
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
{}

...so I removed the kip1patch=nosigchk line.  However, this was a file I had never touched before, so I think this was bundled with one of the hekate downloads.  Should hekate be informed that the .ini file has this setting enabled?

funkyouo, 2019-12-13:
Thank you for your issue. I had a same problem as you when I wanted to boot sysnand to use the Checkpoint. It's just that Atmosphère‘s author has no intention of supporting piracy,you know.