CTCaer/hekate #323: Can't boot into CFW after last update.

AshbalaGang, 2019-12-08:
I'm getting the following error when trying to boot:

Failed to apply 'nosigchk'!
Failed to launch HOS!

I have alredy download the most recent files from Atmosphere and hekate. The payload I'm injecting on is the 5.1.0 and I've updated the libraries.

kristal91, 2019-12-08:
anche io mstesso problema

kristal91, 2019-12-08:
I am also a problem myself

kristal91, 2019-12-08:
I also have this problem

urherenow, 2019-12-08:
Not needed unless you are pirating. Do you think it’s supported on Github?

AshbalaGang, 2019-12-08:
I'm not pirating any type of software.

TKCentaurus, 2019-12-08:
I also only had my previously downloaded games and DLC from the eShop, but I accidentally updated to 9.1.0 while in emuMMC, and i'm getting these same errors.

CTCaer, 2019-12-08:
This is **missing** patches or **broken/mismatched** patches.
I will pretend to believe the no pirate quote, **once**,
and just tell you to remove the `kip1patch=nosigchk`

closed 2019-12-08 by CTCaer