CTCaer/hekate #330: Gen 1 - 9.1.0 fw - Hekate 5.1.1 - Touchscreen not working

Nykoshiro, 2019-12-12:
Original Gen 1 that was auto updated to 9.1.0 fw.  Using 5.1.1 bin via RCM Loader the device boots to Hekate menu.  Temperature & battery status adjust but the device does not respond to touch controls.

The screen works fine when booting to the OFW 9.1.0.  The hardware buttons work when using the (text) menu of Hektate.  I've tried an old Hekate 5.0.2 to test against an older version that I know has worked on another gen 1 device (originally 8.1.0 ofw).

Nykoshiro, 2019-12-12:
The touchscreen also works fine when booting directly to fusee-primary via RCM Loader

Nykoshiro, 2019-12-12:
[image]

The touchscreen test looks similar to issue #274

Nykoshiro, 2019-12-12:
[image]
Another device I have that seems to work without any issues does not exhibit a portion of the screen that is failing to respond to touch.

Nykoshiro, 2019-12-12:
The touchscreen test looks similar to issue #274 

[image]

Nykoshiro, 2019-12-12:
Another device I have that seems to work without any issues does not exhibit a portion of the screen that is failing to respond to touch.
[image]

fennectech, 2019-12-13:
Nah. thats unrelated to #274     that was weirdness with hekate this is almost definitly hardware damage/defect

maskmark, 2019-12-13:
actually it's like this
https://user-images.githubusercontent.com/3665130/62011622-4fd98b80-b183-11e9-9a48-c2d135501104.png

fennectech, 2019-12-13:
Yes. This did not occor in horizon

Nykoshiro, 2019-12-14:
The screen defect only seems to be within Hekate as OFW & CFW work in the areas where Hekate does not.  After 2 days of research & trial/error I was able to finally create nand backups & emummc manually.  I'm not 100% sure if I've replicated the steps that Hekate performs under the GUI but I was finally able to boot my emummc.  Hopefully I did everything right, here's how I did it in the event that it helps someone else with touchscreen issues.  If this is something that doesn't belong here please delete or I can do it.

PC steps
a)  use any modern guide to partition your SD Card accordingly (e.g. 32g on the backend)
b)  copy latest atmosphere, primary.bin & hekate to SD card
c)  rename SDcard:\bootloader\sys\nyx.bin to nyx.bin.bak (this will boot Hekate to text interface)

Switch steps
c)  enter RCM & inject Hekate
d)  Create boot1/2/nand backup > turn off switch > copy your backup to at least 2 other sources

PC steps
e)  Use Hekate combine script or command line to combine your backup bins to a single rawnand.bin (if necessary)
e)  Use MakeEmu.exe (https://github.com/exelix11/MakeEmu/releases) to write boot1/2/rawnand.bin to your emummc partition (be sure to write down your emummc_sector)
f)  Create SDcard:\emummc\emummc.ini file and insert the following:
_[emummc]
enabled=1
sector=<your hex string from makeemu.exe from step e)  for example 0x52F80000>
path=RAW1
id=0x0000
nintendo_path=emummc/Nintendo_

Switch steps
g)  enter RCM & inject Hekate
h)  feel free to set auto boot to fusee-primary (emu) or just boot as normal

CTCaer, 2019-12-14:
This is a duplicate of https://github.com/CTCaer/hekate/issues/274.
Why open a new issue. The tests and all should go there.
Please redirect any comments of that there.

Currently HOS does something that changes the protocol or the calibration and makes it not work in hekate. Funny thing is  that it may start working in a new HOS update as seen by other users. The cause is still unknown.

----------

As for that tool I hope it creates the 16MB protective offset

closed 2019-12-14 by CTCaer

fennectech, 2019-12-14:
In that image you show the issue in HOS

fennectech, 2019-12-14:
> The touchscreen test looks similar to issue #274
> 
> [image]

In this image you show the issue occoring in HOS