CTCaer/hekate #331: fail to init SD card - OFW 9.1.0, hekate to v5.1.1 and Nyx to v0.8.4

dgzhc, 2019-12-14:
Hi,

The switch is under the unpatched group.  XAW70001 series number

Updated the firmware to 9.1.0. Able to entry into RCM. SD is readable from the OFW and able to save snapshot or photo to the SD card. Also, SD works in PC and my friend's switch for jailbreak

After push the hekate_ctcaer_5.1.1 to my switch using Tegra 2.6. Select "launch", then I got the following error "failed to init SD card". 

I did the following things for problem shooting:

1. Tried two different SD cards, one is Samsung EVO 64G and the other one is sandisk 32G. Formatted both in exFat/Fat32 using SDformat and formatted by switch. ---doesn't work, "error -- failed to init SD card"

2. Used switch to format the SD card, save photo and copy photo from/to the SD card. the SD card works fine in OFW. --- SD card works in OFW and PC

3. Used my SD cards with same files in the card in my friend's switch to jailbreak --- can boot to atmosphere without any issues.

4. Reset the whole switch.

any suggestions to resolve this issue?

Thanks very much

CTCaer, 2019-12-14:
This denotes issues with the sd card reader. OFW is able to read that because it supports 1bit mode (1 pin).

dgzhc, 2019-12-15:
> This denotes issues with the sd card reader. OFW is able to read that because it supports 1bit mode (1 pin).

Thanks CTCaer, so the suggestion is to change SD card slot in the nintendo?

CTCaer, 2019-12-15:
The first thing is to check this
[image]
and especially the pin I point to.

dgzhc, 2019-12-15:
thanks CTCaer, I will close this issue and try to fix the problem as you recommended

closed 2019-12-15 by dgzhc