CTCaer/hekate #334: Suggestion: Store current card format

DanielC-08, 2019-12-23:
An idea is when hekate boots is checks the SD card's current filesystem and save it to a file or something like that. This would be helpful for homebrew like game dumping software needing to know the current filesystem

fennectech, 2019-12-24:
I second this idea. Should be simple to impliment too.  we already have all the components there.

CTCaer, 2019-12-31:
Dumping software should use the big file function that automatically splits the files and nothing else.
Anything else is just bad practice. And so I don't see a point on that.
Especially when it can cause issues, since the marked file can be easily copied and mess things up.

TechGeekGamer, 2020-05-13:
Well uh no reason to keep this open I guess..

closed 2020-05-13 by TechGeekGamer