CTCaer/hekate #335: Switch shuts of at 42%

Yukos86, 2019-12-24:
https://github.com/CTCaer/hekate/issues/217#issuecomment-486878618

I have a similar problem where switch shuts of at 42%.
Hekate (v5.0.2) in a battery info says that i have full capacity 8746 mAh instead of 4667mAh.

It seems like "desync" fix not working. How can I fix this problem? Thanks.

fennectech, 2019-12-24:
this doesnt look like a desync problem.  can you show battery info?

Yukos86, 2019-12-24:
[image]

DanielC-08, 2019-12-24:
@Yukos86 Why not update hekate? Not sure if that would change anything, but still is nice to have.

fennectech, 2019-12-24:
Ethere hekate is reading it wrong or something in the BMS is epicly fucked.   Does this happen in HOS too?

Yukos86, 2019-12-25:
Yes, HOS also displays invalid values. The battery life has not changed, so I think that when the HOS shows 42% and the console turns off, the actual charge is about 1-2%

fennectech, 2019-12-25:
idk if the desync bug causes a change in the battery capacity readout.   That is epicly wierd.   can you try running the battery from 0 to 100 to 0 in horizon.  If it doesnt help all you can do is order a replacement battery because your BMS is fucked.

fennectech, 2019-12-25:
replacing your battery wont be difficult.  the hardest part is getting the old one out.

CTCaer, 2019-12-30:
The hekate "desync" fix doesn't fix that.

What you need to do is to remove the battery and any charger, press PWR, wait 5 min, then put it back.
After some full charge cycles, it might fix itself.
This was caused because your battery reached 2520mV.

And you have a working battery but fake percentage. It's not like the old desync issue though. It can still use the whole charge but as I said, it shows wrong %. The old issue didn't allow you to use the full battery.

Lastly, have in mind that here is for reporting issues and feature requests and not whatever support anyone needs. Especially when that has nothing to do with hekate.

closed 2019-12-30 by CTCaer