CTCaer/hekate #336: Inject reboot-to-payload into OFW

lokiuox, 2019-12-28:
As the title says, would it be possible to inject the reboot-to-payload functionality into Stock HOS? It would be nice for people who use dual boot to play online and switch often between CFW and OFW.
Atmosphere already does that, so I was thinking that maybe it could be ported into an injectable patch or somthing. Also, would it be detectable by Nintendo?

CTCaer, 2019-12-30:
Yeah it's possible, but you need a extra stuff from atmosphere. And it needs ams mitm.

closed 2019-12-30 by CTCaer

fennectech, 2019-12-30:
This removes the point of having stock HOS         Would be neat but woudlent be entirely safe.

lokiuox, 2019-12-30:
I'm not sure, but I think it depends on which "level" the patch applies. I mean, it could be simple enough or in such a low level that it COULD be undetectable. It's just a theory though, I just started following the switch scene, so I don't know how HOS works yet, but for sure it's much more subtle than installing an NSP for example.

fennectech, 2019-12-30:
none of us know exactly what can be detected.

lokiuox, 2019-12-30:
From a quick look at ams source code it seems that the reboot-to-payload functionality is implemented into exosphere, the custom trustzone. I wonder if it could work to just load exosphere as the custom trustzone and leave everything else as is to make a "hybrid" OFW. From users feedback, it seems that if you don't install content you don't own and don't launch any homebrews, you don't get banned even if you use a CFW, so theoretically it should be safe.

fennectech, 2019-12-31:
safe is not the same as undetectable.  if nintendo decides to ban for any modifications to the system were all fucked.   modifying OFW is just a plane bad idea

Owen-sys, 2020-07-09:
> Yeah it's possible, but you need a extra stuff from atmosphere. And it needs ams mitm.

you dont have to release this but this could be for banned switches like mines but i didnt want to make this a whole nother issue since its already been done but can you possibly still make this