CTCaer/hekate #338: Hekate Freezes Right After emuMMC creation

GlitchnMoe, 2019-12-30:
Hekate seems to freeze right when tapping the "close" button when it's finished creating the emuMMC. Right when creation of emuMMC was completed, "close" button was tapped. Caused hekate to freeze, needed manual reset to proceed (held power button)

Second attempt made. Created emuMMC to SD partition, waited 10 seconds after receiving completed notification in bottom corner. Same results appeared. Frozen hekate forcing for a manual reset.

Third attempt was made. Created emuMMC to SD partition, waited 1 full minute after receiving completed notification in bottom corner. No freeze and was able to close screen and get back to the hekate home screen. 

battery percentage was roughly 67%-70%, Every time hekate froze, it still was able to create the emuMMC successfully with no sign of corruption to the files.

CTCaer, 2019-12-30:
Thanks for testing 3 times for consistency.
Strange that no one else reported that. It looks like a GUI heap or stack corruption.

It seems you can easily reproduce it (I can't locally), so I will check the code and give you a test binary (which will not write to SD though).

Also have in mind, that the moment you see the Close button light up, the process is finished. So it's strange that hangs there and not before. Did you notice anything else that was weird? Like extra letters or not in the messages.

GlitchnMoe, 2019-12-30:
Not a problem at all. Thank you for responding so fast. 

So to be clear, I would wait essentially until the close button lights up. The first attempt I watched it go from 99% to 100% then hit close immediately after. Thats when it froze. I've created a new emuMMC roughly 3-4 times prior to this instance on the same console and didnt have this problem before. This was the first time this has happened. To be fair though, I dont know how long I waited before hitting the close button in the past.

Before I hit close, nothing out of the ordinary is shown. No extra numbers or letters. Everything looks exactly how it should. No errors of any kind or nothing weird. It just hangs on close with the close button highlighted as it shows I pressed it but stays as if its in a pressed state. No advancement at all.

CTCaer, 2020-03-03:
Please try with the new release because I fixed some heap/list issues.

GlitchnMoe, 2020-03-07:
Yup! That seemed to have fixed it! Tried it last night and pressed close as soon as it lit up. Everything worked as it should. Great job!

closed 2020-03-07 by GlitchnMoe

CTCaer, 2020-03-07:
Thanks for verifying.