CTCaer/hekate #341: Battery de-sync

chrisf95, 2020-01-07:
Hi, my switch at 15% give me the low battery alert and go to stand-by... I try to do a calibration with a fully discharge, but I can't on horizon (because go stand-by every 10 sec) and on hekate menu it shutdown at  about 20%... Furthermore the voltage at a value (about 3000 MV) became a value written in a strange way like "FFF00083000". I don't know what to do because the battery desync fix is automatically done at boot

CTCaer, 2020-01-07:
In hekate, if the voltage drops to 2800mV it shutdowns to protect the battery.
Basically 0% is 3200mV. If it shows 15% or 20% and it's lower or significantly higher, then you have a calibration issue.

This can be normally fixed, with the classic 0-100% (3.2V - 4.2V) full charge cycles.

As for the `FFF8000`, fixed it internally. I forgot to add # which denotes a color value (also the value is wrong, the correct is FF8000 for orange).

chrisf95, 2020-01-07:
I retry and after put it in charge give me 18% on a 2850mv in hekate... So, how I can recalibrate to 3200 = 0%?

P.s. For the FF, I noted that is a FFF8000 with 3 "F", but it's not a problem

CTCaer, 2020-01-07:
Charge it back to 100% (4.2V), then to 0% (3.2V), then back to 100%. **Check note**
This will count as 2 full charge cycles and the % will probably be fixed on its own, or at least get closer to the correct one.

**Note:**
Mind you, that when you reach 100% it's not actually 100%. You need to leave it at 4.2V so it can saturate. Leave it charging for **additional** ~1-2 hours after you see 100% (time depends on the charger used).

chrisf95, 2020-01-07:
Really thanks... I do this yesterday, so today is the 2 full cycle... I hope it resolves

CTCaer, 2020-01-07:
You are welcome.

Also the color will be fixed in next version (https://github.com/CTCaer/hekate/commit/9263e2192fe5190f2381e8d51e9c58062fe867c6).

chrisf95, 2020-01-20:
Hi, I did 2 cycle (100--> 0% at 3.2v) but the 3.2v is marked as 60%. I try to shutdown and power on at 100% in order to apply the fix battery, but nothing... I don't know what do now

CTCaer, 2020-01-20:
Only one power cycle is not enough.
If you don't remove the battery, wait and replug it, it needs some cycles to fix itself.
That's ofc if it wants to fix itself.

chrisf95, 2020-01-20:
I did two cycles, then I continue until It fix. Re-thanks

chrisf95, 2020-01-21:
Today It says 100% all time and in hekate it is 255%... Now it gives me changing value (2% after some second 60, then 100 and this continuously) in also in hekate and booted... What can I do?😱

chrisf95, 2020-01-21:
Today After I put on charge seems all OK, but when it reach 1% while boot, in hekate it is at 3800mv, now I try to discharge in hekate

Online I see a version of hekate with manual battery fix, can this be usefull?

closed 2020-02-29 by CTCaer

chrisf95, 2020-03-17:
Hi, if I recontact you from here but I think is useless open another 3d...
The problem persist: the switch when was shutdown, the next day after I start it says that is 1%

I did different full cycle and I don't know what to do.
online I find a version of hekate with the manual fix, do you think that this can help?
really thanks and sorry for the disturb

CTCaer, 2020-03-18:
It won't help @chrisf95 .
You need to at least try discharging it to 3200mV in hekate. Then charge it fully in HOS. And fully means seeing 100% and leave for about an hour more.

chrisf95, 2020-03-18:
I retry, but I did It many times and when I shutdown and the Day After start It I have 1% also if the battery Is fully charged (4200mv in hekate)

CTCaer, 2020-03-18:
what are the capacity `now` and capacity `full` values?

chrisf95, 2020-03-18:
In this Moment the capacity now Is 0, the full 2930
Sometimes during the full cycle I did It became 4300

CTCaer, 2020-03-18:
Probably HOS has garbage values.
For now the simplest way is to always charge inside HOS until it settles again.

chrisf95, 2020-03-18:
I try thanks

Elionna, 2020-03-19:
I am having a similar issue with the battery desync after not using my switch after 2 months. Before storing it I had autorcm on but I turned off the switch via the power off option in hekate.

Will turning autorcm off help with this problem? I have done 4 cycles and the battery is still shutting off in HOS around 40%, getting worried if I can even fix this.

CTCaer, 2020-03-19:
>For now the simplest way is to always charge inside HOS **until** it settles again.

cristianrj9, 2020-03-31:
hola CTCaer tengo el problema del indicador de bateria de la switch en si la duracion de la bateria es normal el problema es el indicador ya que cuando logre cargalo al 100%  y apagar la consola con ese porcentaje al encenderlo despues de unas horas la consola enciende con el 1% despues de haberlo dejado cargado al 100% como podria solucionarlo?
ayuda por favor