CTCaer/hekate #342: Suggestion: boot when launch is pressed

failsafe42, 2020-01-10:
I think that if there is only one config option in hekate_ipl.ini, when you press launch it should just boot that option instead of going into the config selection screen.

CTCaer, 2020-01-10:
Hekate supports auto boot of a boot entry since ever.

Go to options-> auto boot -> select what you want to automatically boot -> hit the save options button on the bottom.
It basically skips the whole procedure of going into the menu.
And you can enter in it again by pressing VOL- when the boot logo is displayed.

You can even change the wait time for bootlogo (it also affects how long it waits for VOL- in case you want to enter the menu again)

closed 2020-01-10 by CTCaer

failsafe42, 2020-01-10:
I don't want to completely bypass hekate, I just want the launch button to automatically choose the config.

CTCaer, 2020-01-10:
Well I don't see a point to that, but anyway, new version will have something else that might work for your use case

fennectech, 2020-01-10:
I do like the idea of a “boot autoboot” button on the home screen that jsut boots whats currently set as autoboot but the ammount of convience it adds would be minimal.